arduino/Arduino #9407: The download version of Arduino does not work with newest macOS update

hanne-louise, 2019-11-07:
I don't know if it is a bug but with macOS Catalina 10.15 I can not open the download version of Arduino

gvarisco, 2019-11-11:
Hi @hanne-louise, could you please tell us what is the errors that gets displayed and/or attach a screenshot?

aurman, 2019-11-16:
“Arduino” can’t be opened because Apple cannot check it for malicious software.

This software needs to be updated. Contact the developer for more information.

[image]

per1234, 2019-11-18:
Closing as duplicate of https://github.com/arduino/Arduino/issues/9314

closed 2019-11-18 by per1234