arduino/Arduino #9411: RS485 Getting Started Doc

fldemarco, 2019-11-08:
For the Arduino RS485 Shield, on the Getting Started page the 3 DIP switch position functions seem backwards after studying the schematics provided.
Can someone verify this is true?
Thanks.

graydonhope, 2020-03-16:
Can you please link the page you are referring to? 

> position functions seem backwards?

Can you elaborate on this? What do you mean backwards?

[Getting Started with the MKR 485 Shield](https://www.arduino.cc/en/Guide/MKR485Shield) is this the page you are referring to?

fldemarco, 2020-03-16:
Yes sir, that's the document.  Back in Nov. I thought the positions of the DIP switches were inverted - the function listed as the OFF position should have been listed under the ON position and vice versa.  This was based on trying to follow this schematic: https://content.arduino.cc/assets/MKR485ShieldV3.0_sch.pdf 

But maybe it's my ability to read the schematic.  And I had to abandon the study of this board.  I do plan to pick it up again sometime in the future.

Thanks.
Frank

graydonhope, 2020-03-23:
@fldemarco I have looked into this issue and see what you are saying. On the MKR 485 shield [schematics](https://content.arduino.cc/assets/MKR485ShieldV3.0_sch.pdf) you will notice they used the MAX3157 microcircuit. I found the MAXIM MX3157 [datasheet](https://datasheets.maximintegrated.com/en/ds/MAX3157.pdf) which includes the pin descriptions and many other useful operating characteristics / schematics that you may wish to use in the future. 

Referencing page 11 from the MX3157 schematic, it does appear that switch 2 is reversed. When switch 2 is ON, it makes a connection to pin 14 IGND which is ISOCOM on the MX3157 circuit. According to their description, pin number 10 is the Half-Duplex, Full-Duplex selector and when connected to ISOCOM, it is in Full-Duplex mode. 

This is their description for pin 10:
> Half/Full-Duplex Selector Pin. Leave open, or connect to ISOCOM to select Full Duplex, orconnect to ISOVCC to select Half Duplex.

I will continue to look into the circuit and read the MX3157 datasheet more so I can provide you with a more detailed explanation and ensure we have not missed anything. If you haven't already done so, I suggest looking at the [MX3157 datasheet](https://datasheets.maximintegrated.com/en/ds/MAX3157.pdf) as well.

fldemarco, 2020-03-23:
OK sir. I will. Thanks. Frank

closed 2020-09-29 by dcuartielles