arduino/Arduino #9412: dont want to open

Macka32, 2019-11-08:
When i open arduino ide the loading screen shows for 2 seconds and then it closes,
i tryed to reinstall it from the microsoft store, but its still crashing.Then installing from the arduino website but its the same

matthijskooijman, 2019-11-08:
Could you try running `arduino_debug.exe` (instead of `arduino.exe`)? It should be in the same folder and might show some more information that helps to diagnose this. You might need to run it from a command prompt, though.

Macka32, 2019-11-10:
I am using Microsoft store version of arduino ide

per1234, 2021-04-03:
Closing because there is not enough information to enable an investigation and enough time has passed that the issue is no longer of value for tracking occurrences.

@Macka32 if you are still experiencing this issue with the latest version of the Arduino IDE, you're welcome to comment here and I'll investigate.

closed 2021-04-03 by per1234