arduino/Arduino #9419: Can add realtime port selection to "Preference.txt"?

xidameng, 2019-11-11:
I am using a third-party program to build the binary file and upload it under Arduino IDE's GUI, meaning I am not using the built-in avr-gcc nor avrdude for building and uploading. As there is a on-board DAP chip, uploading binary file is done by copy&paste the binary file into the flash memory recognized as a removable storage by the OS.

However, this make reading user's current port selection very difficult because the IDE is written in Java, but the thrid-party program I am using is in C++. 

Knowing which port the user has currently selected is very important to me, as sometimes I need to program multiple boards at the same time and monitor their received data on the serial port.

If someone can point me into the right direction or link me to the previously-solved similar question , it will be much appreciated!

PS: I am working under Ubuntu OS and using latest Arduino IDE.

matthijskooijman, 2019-11-11:
I'm not sure I understand your question completely, but can you not simply pass the selected port to your program using a commandline argument? This is also how the port is passed to avrdude?

xidameng, 2019-11-13:
> I'm not sure I understand your question completely, but can you not simply pass the selected port to your program using a commandline argument? This is also how the port is passed to avrdude?

but how? which command lines am I supposed to use to pass the port to my uploading program?

matthijskooijman, 2019-11-13:
Here's how it works for avrdude (note the `"-P{serial.port}"`): https://github.com/arduino/ArduinoCore-avr/blob/c8d6aef6d9331af1ec8dea6f78b4e43174cdb55d/platform.txt#L94-L105

See also https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5-3rd-party-Hardware-specification for more elaborate documentation.

xidameng, 2019-11-13:
> Here's how it works for avrdude (note the `"-P{serial.port}"`): https://github.com/arduino/ArduinoCore-avr/blob/c8d6aef6d9331af1ec8dea6f78b4e43174cdb55d/platform.txt#L94-L105
>
> See also [...]

Thanks matthijskooijman!
I will look into the platform.txt file, and see if the selected serial port can be passed into my uploading program as it does with avrdude :)
You have a good day!

matthijskooijman, 2019-11-13:
Good luck! I'm going to close this issue for now, if you have further remarks you can leave them here (or maybe ask on the Arduino forums, which is a more appropriate place for questions than here, really).

closed 2019-11-13 by matthijskooijman