arduino/Arduino #9420: Can't this only use C++?

deepnum, 2019-11-11:
It would be nice if no Java is used.

matthijskooijman, 2019-11-11:
I'm not entirely sure what you're suggesting here? Is this a suggestion to rewrite the entire IDE in C++ rather than java? That does not seem like an appropriate suggestion for this repository (and that seems unlikely to happen as well). Maybe you mean something completely different, but in that case you should be more specific about what you mean.

I'll go ahead and close this issue, since it does not seem useful right now. If you want to add more information, we can re-open later if needed.

closed 2019-11-11 by matthijskooijman

deepnum, 2019-11-13:
@matthijskooijman my suggestion is to rewrite the entire IDE in C++.