arduino/Arduino #9427: Issues in WiFiNiNa for mkr 1010 and Nano 33 IoT

racpra16, 2019-11-14:
2 functions called setupNetwork(), you can change between AP & connecting to existing network

If connect to an existing network. It can reach the host using trace root and the hostname. 

If creating an Ap and connect to the create ap. It cannot reach, the arduino using trace root and the hostname. But he can using an Arduino AP. 

Parallelly, hostByName function works on the network but not on the AP. Also, the function setHostName does not seem to work on  the Arduino nano IOT 33

[Re wifinina library.pdf](https://github.com/arduino/Arduino/files/3846693/Re.wifinina.library.pdf)
[from Russell(1).zip](https://github.com/arduino/Arduino/files/3846696/from.Russell.1.zip)

sandeepmistry, 2019-11-14:
@racpra16 @Rocketct this should be "moved" to: https://github.com/arduino-libraries/WiFiNINA/issues

Russell108, 2019-11-15:
Hi Since this was created by one of your support members
would it not help to move it to the right place. And save me time & frustration
I am Russell who first contacted the support team.
This is a big issue for me & if not fixed I need to consider using different product
I purchased nano BLE but that had issues with the BLE library meaning I could not use it
is it going to be the same with nano iot 33

Thank you Anyways
=================================================================

Rachit Prabhakar (Arduino) Nov 14, 17:08 CET 
Dear Russell, Thank you for contacting us. We have pushed your issue on the Github repo of Arduino libraries. The developers will be in direct contact with you on the GitHub. Also, I humbly request you to open new threads(issues) on GitHub repo of Arduino libraries for debugging your query by the developers. Moreover, for the static IP issue using config() functions, please elaborate it more with the sketch so that developers can debug your query more easily. Kind Regards, Rachit Bandhu Prabhakar
--

==================================================================

Rocketct, 2019-11-15:
Hi @Russell108 nice to meet you, sorry but this should be part of the  https://github.com/arduino-libraries/WiFiNINA/issues this give visibility to all the user that have the same problem, you will receive an e-mail each time the issue will be commented.
Saw this, i have just reproduced the issue, in my workbench, i have:
- 1 mkr1010 that works as access point with a assigned ip like 123.123.12.12
- 1 nano iot 33 that connect access to this access point and try to resolve the host name as an ip.
Please add comment if this is the  wrong configuration.

Configured the boards i found two scenarios:
1 Scenario:  if you consider the Example->WiFiNINA->AP_SimpleWebServer it say to check on httt://123.123.12.12/ the behavior of the AP if you give this name like follow:

``` 
  char server[] = "123.123.12.12"; 
  IPAddress result;
  int err = WiFi.hostByName(server, result) ;
  if(err == 1){
        Serial.print("Ip address: ");
        Serial.println(result);
  } else {
        Serial.print("Error code: ");
        Serial.println(err);
  }
```
 to the iot 33 it resolve correctly the ip and all goes fine, this formally is different from give an object of the type `IPAddress` because the library call the API to resolve the URL through its DNS as the correct IPAddress. 
 
2 Scenario: using setHostName, this not works, in first analysis, this is because nina f/w side there is not the support for the name assignment, but i have also tested the default name `arduino-mac[4]mac[5]`, where mac[4]mac[5] are the last two number of the board's mac address, now i'm follow to investigate over this and if a patch will available i'll make a PR.

Another important consideration is that if you simply test the piece of code i share with the iot 33 connected to any classic AP the name is correctly resolved.
this are all the results of my test.

closed 2019-11-15 by per1234