arduino/Arduino #9431: "#pragma region" and "#pragma endregion" not possible with compiler warning on Arduino IDE

wrgallo, 2019-11-15:
Bug reproduction steps:

1. Arduino 1.8.10 IDE > File > Preferences > Set Compiler Warnings to "More" or "All"

2. In your code divide your region with "#pragma region NAME_OF_REGION" and "#pragma endregion"

Example:
```
#pragma region CONSTANTS
const int myConstant = 3;
#pragma endregion // CONSTANTS
```

3. Try to Verify/Upload

Expected Result:

- #pragma region , #pragma endregion won't be treated as unknown pragmas

Actual Result:

- The warning of the unknown pragma region, and endergion will be treated as errors ([-Werror=unknown-pragmas]).

------------
Pragma region is very popular and useful to collapse regions with others editors such Visual Studio Code. (https://docs.microsoft.com/en-us/cpp/preprocessor/region-endregion?view=vs-2019)

per1234, 2019-11-18:
Which board do you have selected from the **Tools > Board** menu @wrgallo?

I can't reproduce this when compiling with any of the official Arduino cores. My suspicion is that you are compiling with a 3rd party core such as ESP32 that promotes warnings to errors via `-Werror=all` (https://github.com/espressif/arduino-esp32/blob/1.0.4/platform.txt#L20-L21).

wrgallo, 2019-11-18:
> Which board do you have selected from the **Tools > Board** menu @wrgallo?
>
> I can't reproduce this when compiling with any of the official Arduino cores. My suspicion is that you are compiling with [...]

Yes, I was using a ESP32 board, my mistake.

closed 2019-11-18 by wrgallo