arduino/Arduino #9433: 1.8.11

victorbramhall, 2019-11-16:
Have always downloaded the latest Nightly version.  Since the last version in October, the IDE is void of text. I haven't changed any Windows 10 settings.
[image]

facchinm, 2019-11-20:
I think it might be related to the new  Java runtime (adoptJDK) taht is being bundled in the hourly builds...
Can you try launching `arduino_debug.exe` and paste here all the output? Thanks!

victorbramhall, 2019-11-20:
Set log4j store directory C:\Users\victo\AppData\Local\Arduino15
Loading configuration...
Initializing packages...
DEBUG StatusLogger Using ShutdownCallbackRegistry class org.apache.logging.log4j.core.util.DefaultShutdownCallbackRegistry
DEBUG StatusLogger Not in a ServletContext environment, thus not loading WebLookup plugin.
DEBUG StatusLogger AsyncLogger.ThreadNameStrategy=UNCACHED (user specified null, default is UNCACHED)
TRACE StatusLogger Using default SystemClock for timestamps.
DEBUG StatusLogger org.apache.logging.log4j.core.util.SystemClock does not support precise timestamps.
DEBUG StatusLogger Not in a ServletContext environment, thus not loading WebLookup plugin.
DEBUG StatusLogger Took 0.083283 seconds to load 209 plugins from sun.misc.Launcher$AppClassLoader@af3868
DEBUG StatusLogger PluginManager 'Converter' found 44 plugins
DEBUG StatusLogger Starting OutputStreamManager SYSTEM_OUT.false.false-1
DEBUG StatusLogger Starting LoggerContext[name=af3868, org.apache.logging.log4j.core.LoggerContext@57d4dd]...
DEBUG StatusLogger Reconfiguration started for context[name=af3868] at URI null (org.apache.logging.log4j.core.LoggerContext@57d4dd) with optional ClassLoader: null
DEBUG StatusLogger Not in a ServletContext environment, thus not loading WebLookup plugin.
DEBUG StatusLogger PluginManager 'ConfigurationFactory' found 4 plugins
DEBUG StatusLogger Not in a ServletContext environment, thus not loading WebLookup plugin.
DEBUG StatusLogger Not in a ServletContext environment, thus not loading WebLookup plugin.
DEBUG StatusLogger Missing dependencies for Yaml support, ConfigurationFactory org.apache.logging.log4j.core.config.yaml.YamlConfigurationFactory is inactive
DEBUG StatusLogger Not in a ServletContext environment, thus not loading WebLookup plugin.
DEBUG StatusLogger Not in a ServletContext environment, thus not loading WebLookup plugin.
DEBUG StatusLogger Using configurationFactory org.apache.logging.log4j.core.config.ConfigurationFactory$Factory@1919693
TRACE StatusLogger Trying to find [log4j2-testaf3868.properties] using context class loader sun.misc.Launcher$AppClassLoader@af3868.
TRACE StatusLogger Trying to find [log4j2-testaf3868.properties] using sun.misc.Launcher$AppClassLoader@af3868 class loader.
TRACE StatusLogger Trying to find [log4j2-testaf3868.properties] using sun.misc.Launcher$AppClassLoader@af3868 class loader.
TRACE StatusLogger Trying to find [log4j2-testaf3868.properties] using ClassLoader.getSystemResource().
TRACE StatusLogger Trying to find [log4j2-testaf3868.yml] using context class loader sun.misc.Launcher$AppClassLoader@af3868.
TRACE StatusLogger Trying to find [log4j2-testaf3868.yml] using sun.misc.Launcher$AppClassLoader@af3868 class loader.
TRACE StatusLogger Trying to find [log4j2-testaf3868.yml] using sun.misc.Launcher$AppClassLoader@af3868 class loader.
TRACE StatusLogger Trying to find [log4j2-testaf3868.yml] using ClassLoader.getSystemResource().
TRACE StatusLogger Trying to find [log4j2-testaf3868.yaml] using context class loader sun.misc.Launcher$AppClassLoader@af3868.
TRACE StatusLogger Trying to find [log4j2-testaf3868.yaml] using sun.misc.Launcher$AppClassLoader@af3868 class loader.
TRACE StatusLogger Trying to find [log4j2-testaf3868.yaml] using sun.misc.Launcher$AppClassLoader@af3868 class loader.
TRACE StatusLogger Trying to find [log4j2-testaf3868.yaml] using ClassLoader.getSystemResource().
TRACE StatusLogger Trying to find [log4j2-testaf3868.json] using context class loader sun.misc.Launcher$AppClassLoader@af3868.
TRACE StatusLogger Trying to find [log4j2-testaf3868.json] using sun.misc.Launcher$AppClassLoader@af3868 class loader.
TRACE StatusLogger Trying to find [log4j2-testaf3868.json] using sun.misc.Launcher$AppClassLoader@af3868 class loader.
TRACE StatusLogger Trying to find [log4j2-testaf3868.json] using ClassLoader.getSystemResource().
TRACE StatusLogger Trying to find [log4j2-testaf3868.jsn] using context class loader sun.misc.Launcher$AppClassLoader@af3868.
TRACE StatusLogger Trying to find [log4j2-testaf3868.jsn] using sun.misc.Launcher$AppClassLoader@af3868 class loader.
TRACE StatusLogger Trying to find [log4j2-testaf3868.jsn] using sun.misc.Launcher$AppClassLoader@af3868 class loader.
TRACE StatusLogger Trying to find [log4j2-testaf3868.jsn] using ClassLoader.getSystemResource().
TRACE StatusLogger Trying to find [log4j2-testaf3868.xml] using context class loader sun.misc.Launcher$AppClassLoader@af3868.
TRACE StatusLogger Trying to find [log4j2-testaf3868.xml] using sun.misc.Launcher$AppClassLoader@af3868 class loader.
TRACE StatusLogger Trying to find [log4j2-testaf3868.xml] using sun.misc.Launcher$AppClassLoader@af3868 class loader.
TRACE StatusLogger Trying to find [log4j2-testaf3868.xml] using ClassLoader.getSystemResource().
TRACE StatusLogger Trying to find [log4j2-test.properties] using context class loader sun.misc.Launcher$AppClassLoader@af3868.
TRACE StatusLogger Trying to find [log4j2-test.properties] using sun.misc.Launcher$AppClassLoader@af3868 class loader.
TRACE StatusLogger Trying to find [log4j2-test.properties] using sun.misc.Launcher$AppClassLoader@af3868 class loader.
TRACE StatusLogger Trying to find [log4j2-test.properties] using ClassLoader.getSystemResource().
TRACE StatusLogger Trying to find [log4j2-test.yml] using context class loader sun.misc.Launcher$AppClassLoader@af3868.
TRACE StatusLogger Trying to find [log4j2-test.yml] using sun.misc.Launcher$AppClassLoader@af3868 class loader.
[... cut ...] dfront], Server=[cloudflare], CF-Ray=[538aa224eb0558e9-DFW], Content-Range=[bytes 0-281218/281219], X-Amz-Cf-Pop=[DFW53-C1], Connection=[keep-alive], Last-Modified=[Wed, 13 Nov 2019 09:48:55 GMT], x-amz-version-id=[IL8l1Bg7.qbdZ2Ph.08WFcjeokCBEbvC], Date=[Wed, 20 Nov 2019 13:05:04 GMT], Via=[1.1 256565adef2758225f5fdec519d8ff68.cloudfront.net (CloudFront)], x-amz-replication-status=[PENDING], CF-Cache-Status=[HIT], Cache-Control=[max-age=3600], ETag=["3a5dac37c8c64dce393fe6a7b2876d42"], X-Amz-Cf

facchinm, 2019-11-20:
There's nothing related with UI stuff... Can you try temporarily removing `package_samd-hourly-build_index.json` from the additional board manager urls?

victorbramhall, 2019-11-20:
Yes, I'll do that.

Thank you for your help.

victorbramhall, 2019-11-20:
I normally unzip the install file and just replace the files that have changed. That has worked up until now. I deleted the folder and did a fresh unzip. The program runs just fine now. Thank you for the help!

facchinm, 2019-11-20:
Probably the fact that JRE has changed triggered the issue. Glad it is solved! Closing as fixed

closed 2019-11-20 by facchinm