arduino/Arduino #9436: Please add TFTLCD-Library to the library manager

dherrada, 2019-11-17:
Please add this repository:
https://github.com/adafruit/TFTLCD-Library

per1234, 2019-11-18:
You need to create a [release](https://help.github.com/articles/creating-releases/) or [tag](https://git-scm.com/docs/git-tag) that matches the `version` value in your library.properties file.

After doing so, please comment here to let us know we can proceed with adding your library to the Library Manager index.

For more information see:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

dherrada, 2019-12-02:
@per1234 Just added the release. Should be all good now.

dherrada, 2019-12-05:
@per1234 So sorry to bug you. It'd be really great if this could get added soon. I'm waiting on this to be added to continue with a project I've been assigned. Thanks!

per1234, 2019-12-06:
Hi @dherrada. I don't have access to add libraries. I just label the issues so that they can be easily organized. cmaglie occasionally comes through to clear all the outstanding Library Manager inclusion requests and closes those issues when that is done. You will receive a notification from GitHub of this and your library will be available in Library Manager shortly after that. So we'll get to it soon, please be patient.

dherrada, 2019-12-07:
@per1234 I understand. Thanks for explaining that to me.

cmaglie, 2019-12-17:
Done, sorry for the long delay

closed 2019-12-17 by cmaglie

dherrada, 2020-01-12:
@cmaglie Thank you