arduino/Arduino #9438: Cannot program ESP32 with macOS Catalina

praeclarum, 2019-11-17:
After updating to Catalina and ESP32 1.0.4 I cannot build and deploy. The only error I get is:

```
fork/exec /Applications/Arduino.app/Contents/Java/tools-builder/ctags/5.8-arduino11/ctags: bad CPU type in executable
Error compiling for board ESP32 Wrover Module.
```

This error was previously discussed in #8974

facchinm, 2019-11-20:
On Catalina you **need** to use IDE 1.8.10, otherwise the 32bit tools will be discarded and the compilation will fail.

closed 2019-11-20 by facchinm

shaxinwei, 2021-03-06:
My Mac OS X (Mojave) does not seem to see my M5Stick-C as a serial device.  I tried two different M5Stick-C's.   In fact my students have had the same problem .   Arduino-1.8.10.app IDE Too/s / Port:… menu lists only "/dev/cu.Bluetooth-Incoming-Port"  and not the M5Stick-C. 

So flashing the M5Stick-C yields only the following error:

> Arduino: 1.8.10 (Mac OS X), Board: "M5Stick-C, Default, 115200, None"
> 
> Sketch uses 708241 bytes (54%) of program storage space. Maximum is 1310720 bytes.
> Global variables use 39928 bytes (12%) of dynamic memory, leaving 287752 bytes for local variables. Maximum is 327680 bytes.
> esptool.py v3.0-dev
> Serial port /dev/cu.Bluetooth-Incoming-Port
> Connecting........_____....._____....._____....._____....._____....._____.....____An error occurred while uploading the sketch
> _
> 
> A fatal error occurred: Failed to connect to ESP32: Timed out waiting for packet header
> 

Any tips?