arduino/Arduino #9439: [Library Manager] ezButton Library

ArduinoGetStarted, 2019-11-17:
Please add the ezButton Library to Library Manager: [https://github.com/ArduinoGetStarted/button](https://github.com/ArduinoGetStarted/button)

ArduinoGetStarted, 2019-11-19:
It has been 16 hours since the library is added. But I cannot search it in library manager in Arduino IDE. I would like to ask whether there is any problem?

per1234, 2019-11-19:
@ArduinoGetStarted your library was not added. I simply labeled the issue in preparation for the bulk add that will happen sometime within the next couple of weeks.

Your library will not be available for installation via Library Manager until it has been added to the index. cmaglie occasionally comes through to clear all the outstanding Library Manager inclusion requests and closes those issues when that is done. You will receive a notification from GitHub of this and your library will be available in Library Manager shortly after that. So we'll get to it soon, please be patient.

ArduinoGetStarted, 2019-11-19:
I got it. Thank you for quick reply

cmaglie, 2019-11-25:
Done.

closed 2019-11-25 by cmaglie