arduino/Arduino #9444: [Library Manager] Add Library LibSemaforo

javisank, 2019-11-18:
Please add https://github.com/javisank/LibSemaforo

Thanks!

per1234, 2019-11-19:
In order to be added to the Library Manager index your library must contain a library.properties file. The format of this file is documented here:
https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5:-Library-specification#libraryproperties-file-format

After adding the library.properties file, you will need to create a new tag/release or redo the existing one. Make sure that the `version` value matches the tag of the release that contains it.

Once you have done this please comment here so we know we can proceed.

For more information see:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

javisank, 2019-11-19:
Thanks for the quick response. I must have forgotten to do the commit with the library.properties file. I have just done it. 
Is there anything else to correct?

per1234, 2019-11-19:
@javisank everything looks good to me now. Thanks!

cmaglie, 2019-11-25:
Done.

closed 2019-11-25 by cmaglie