arduino/Arduino #9445: [Library Manager] Cerebro educational robot

cerebro11101, 2019-11-18:
https://github.com/cerebro11101/cerebro

per1234, 2019-11-19:
You need to create a [release](https://help.github.com/articles/creating-releases/) or [tag](https://git-scm.com/docs/git-tag) that matches the `version` value in your library.properties file.

After doing so, please comment here to let us know we can proceed with adding your library to the Library Manager index.

For more information see:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

cerebro11101, 2019-11-19:
sorry :) its ready, !

cerebro11101, 2019-11-19:
> You need to create a [release](https://help.github.com/articles/creating-releases/) or [tag](https://git-scm.com/docs/git-tag) that matches the `version` value in your library.properties file. [...]

sorry :) its ready, !

cmaglie, 2019-11-25:
Done.

closed 2019-11-25 by cmaglie