arduino/Arduino #9447: Mingw-w64 runtime failure

italocjs, 2019-11-19:
Hi, i'm unable to compile any sketch, even examples, already tried reinstalling but didnt work:

```

Mingw-w64 runtime failure:

  VirtualProtect failed with code 0x5afavr-g++: internal compiler error: Aborted (program cc1plus)

Please submit a full bug report,

with preprocessed source if appropriate.

See <https://gcc.gnu.org/bugs/> for instructions.

"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -MMD -flto -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10810 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\cores\\arduino" "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\variants\\mega" "C:\\Users\\italo\\AppData\\Local\\Temp\\arduino_build_227261\\sketch\\src\\HAL\\shared\\backtrace\\unwarm_thumb.cpp" -o "C:\\Users\\italo\\AppData\\Local\\Temp\\arduino_build_227261\\sketch\\src\\HAL\\shared\\backtrace\\unwarm_thumb.cpp.o"
exit status 4
Erro compilando para a placa Arduino/Genuino Mega or Mega 2560

```

Anyone has any idea? i tried googling the error but didnt found anything helpful for windows

facchinm, 2019-11-20:
Hi @italocjs ,
you are the first person reporting this king of issue... Can you post the specifications of your Windows installation? Thanks!

italocjs, 2019-11-20:
Hi, i believe it was related to RAM being used at max, i monitored the task manager and the windows defender went insane, it was consuming ~40% of cpu, and 6gb of ram alone (i only have 8) when the compiling process started, after disabling the windows the defender i was able to compile again.

The weird thing is that this happened even with the blink example.

About the laptop, 
Asus X510UAR, 256gb ssd, 8gb ram, win10 pro, no graphic card.

closed 2019-11-20 by italocjs