arduino/Arduino #9456: [Library Manager] Please Yesbotics libraries

kappaj, 2019-11-23:
Please add:

https://gitlab.com/yesbotics/libs/arduino/voltmeter
https://gitlab.com/yesbotics/libs/arduino/interval
https://gitlab.com/yesbotics/libs/arduino/average-value
https://gitlab.com/yesbotics/libs/arduino/timeout

Thanks!

per1234, 2019-11-23:
Hi @kappaj. Each library added to the Library Manager index is required to have a unique `name` value in library.properties. Unfortunately, "Timeout" is already taken. In order to add this library to the Library Manager index, you will need to change the value of the `name` field in library.properties to a unique name, then either create a new tag or else redo the existing tag. If you are creating a new tag, make sure to update the `version` field in library.properties before creating the tag.

Your other libraries look fine to me.

Once you have done this, please comment here so we'll know we can proceed. Thanks!

kappaj, 2019-12-19:
Hi @per1234,
I updated my libs and renamed the interval and timeout libs to IntervalCallback and TimeoutCallback. I also renamed this twi gitlab repos and tagged it. Theses are the final repos:

https://gitlab.com/yesbotics/libs/arduino/voltmeter
https://gitlab.com/yesbotics/libs/arduino/interval-callback
https://gitlab.com/yesbotics/libs/arduino/average-value
https://gitlab.com/yesbotics/libs/arduino/timeout-callback

Thank you!

cmaglie, 2019-12-29:
Done.

closed 2019-12-29 by cmaglie