arduino/Arduino #9460: Arduino fails to launch on a Mac OS 10.15

PJOssenbruggen, 2019-11-24:
I found a post that states an _Arduino will successfully open by right clicking_. In my case, it initializes, starts and immediately closes. 

I am suspicious. The Arduino app did not load properly. I got the following message. I found that in System Prefences Security & Preference would not open. I fixed this "malicious software" problem by removing a cache associated with Security & Preference and rebooting my Mac in a safe mode. Pressing the shift key when booting. After doing this, I was sort of able to open Arduino. See paragraph one above.

Apparently, I need to do something more than this.

Help will be appreciated.

[image]

EvgenV, 2019-11-24:
Hi. 
I have had  the same issue. Try to open System Preferences -> Security and Privacy and press button Open Anyway. It helped me
[image]

per1234, 2019-11-24:
Closing as duplicate of https://github.com/arduino/Arduino/issues/9314

closed 2019-11-24 by per1234

PJOssenbruggen, 2019-11-25:
Can Arduino run on Mac OSX 10.15 Catalina?

PR #10357 (2020-06-12): Updated appbundler to 1.0ea-arduino8 (fix crash on Macosx Catalina + multi monitor setup)
As discussed in #10234 building Appbundler against MacOSX SDK 10.9 solved the crash on macosx catalina + multi-monitor setup.

Supersedes #10234
Fix #9926 
Fix #9817 
Fix #10302 
Fix #10283 
Fix #10104 
Fix #9957 
Fix #9828 
Fix #9460