arduino/Arduino #9462: Serial Plotter - support a comment character to ignore incoming text lines

AESilky, 2019-11-24:
Serial Plotter added the ability to supply labels for the data as a comma/tab/space separated line of text. The ability to indicate that a text line should be ignored (not interpreted as labels) would help keep labels from being added when not wanted while being able to print debugging information to the Serial Monitor.

Something like a leading '!' or some other character:  
 ! This will not be interpreted as data labels, even though it has spaces and a comma.

madsdyd, 2020-04-07:
AFAICT, it is trivial to add something hardcoded to the Serial Plotter, as it is. This patch makes the Serial Plotter ignore lines starting with #:

```diff
diff --git a/app/src/processing/app/SerialPlotter.java b/app/src/processing/app/SerialPlotter.java
index 035005a..366bc57 100644
--- a/app/src/processing/app/SerialPlotter.java
+++ b/app/src/processing/app/SerialPlotter.java
@@ -401,8 +401,8 @@ public class SerialPlotter extends AbstractMonitor {
       messageBuffer.delete(0, linebreak + 1);
 
       line = line.trim();
-      if (line.length() == 0) {
-        // the line only contained trimmable characters
+      if (line.length() == 0 || line.startsWith("#")) {
+        // the line only contained trimmable characters or should be ignored
         continue;
       }
       String[] parts = line.split("[, \t]+");
```

However, I would suggest adding a regular expression to the GUI instead: Any lines matching a specific regular expression gets interprented by the Serial Plotter. Default value: "^.+$". 
This would allow the user to filter lines based on need. 

I am sorely lacking in Swing experience though, otherwise I would not mind giving the implementation a try.

A question: Is there any canonical documentation for the Serial Plotter? I had a look through the source code yesterday to use it correctly, and would not mind documenting the existing performance. Guides on the net are inadequate.

per1234, 2020-04-07:
>Is there any canonical documentation for the Serial Plotter?

There is this: https://github.com/arduino/Arduino/blob/master/build/shared/ArduinoSerialPlotterProtocol.md

The only documentation other than that is the demonstration of using Serial Plotter in some of the tutorials related to the example sketches, as listed here: https://github.com/arduino/Arduino/pull/7453#issuecomment-469207719

madsdyd, 2020-04-07:
> > Is there any canonical documentation for the Serial Plotter?
> 
> There is this: https://github.com/arduino/Arduino/blob/master/build/shared/ArduinoSerialPlotterProtocol.md

Ah, thanks! I was completely unable to locate this information. My google-foo was inadequate. 

And thank you for pointing me to some of the discussion about the Serial Plotter. Hugely useful tool, that could be even more useful with just a few additions. Do you think a PR adding a "selector" based on regular expressions to the GUI would be accepted?

rnestler, 2020-05-18:
> AFAICT, it is trivial to add something hardcoded to the Serial Plotter, as it is. This patch makes the Serial Plotter ignore lines starting with #:

For most use cases this would probably be enough. It is easy to just prefix log messages with `#`.

Batwam, 2021-03-06:
hi everyone, has anyone found a solution to exclude some outputs from the Serial plotter without having to patch the java file?

stefan123t, 2021-05-06:
@per1234 the other issue in #10226 can be marked as duplicate.

please double read the explanation of @madsdyd here:

However, I would suggest adding a regular expression to the GUI instead:
**Any lines matching a specific regular expression gets interpreted** by the Serial Plotter.
Default value: "^.+$".
This would allow the user to filter lines based on need.