arduino/Arduino #9469: Uno WiFi Rev2 WiFi port bricked by Firmware updater

jwalters1955, 2019-11-25:
IDE: Arduino 1.8.8 running on network booted Raspberry Pi 3 B+.

Update of WiFinina firmware from 1.0.0 (as supplied) to 1.2.1.

Running CheckFirmwareVersion returned V1.0.0.

Loaded "FirmwareUpdater" sketch and ran the "WiFi101 / WiFiNINA Firmware updater" plugin.

Selected "Arduino Uno WiFi Rev2(/dev/ttyACM0)" and  "NINA firmware(1.2.1)(Arduino Uno WiFi Rev2)"

"Test Connection" reports that the programmer is working.

"Update Firmware" erases the target and programs 880640 bytes. This takes about two minutes. The plugin then reports  "Error validating flashed firmware".

The console reports;
java.lang.Exception: Error validating flashed firmware.
        at cc.arduino.plugins.wifi101.flashers.java.Ninaflasher.updateFirmware(NinaFlasher.java:106)
        at cc.arduino.plugins.wifi101.UpdaterImpl$2.run(Updater.Impl.java:196)

"CheckFirmwareVersion" now reports "Communication with WiFi module failed" although reloading the updater sketch and "Test Connection" still reports "Programmer is working"

I've re-run the process a couple of times but to no avail.

WiFi processor effectively bricked!!!

facchinm, 2019-11-29:
Hi @jwalters1955 ,
the problem is related to the timeout (2 minutes) which is a bit too tight for UNO wifi rev2 communication speed.
The fix has been included in the firmware updater shipped with IDE 1.8.10 . 
May you try it and report if you can "unbrick" the board? Thanks!

djmastrianni, 2020-01-25:
I have the same problem but using IDE 1.8.10 as suggested above.  I actually had everything working but when I ran the Arduino supplied CheckfirmwarVersion sketch, it kept failing the firmware versions comparison.  So I made the terrible mistake of running the supplied firmware updater and now also have a bricked WiFi board.  If I can't  restore the firmware that came installed with the UNO WiFi Rev2 soon, I will have to send it back and try to get a refund.

jwalters1955, 2020-01-27:
panic not!! I still can't get 1.8.10 to work under Linux but it appears to work OK under Win 10. I used the inbuilt sketch under 1.8.10 to re-flash the WiFi module and it worked fine.

djmastrianni, 2020-01-27:
It seems that if it’s working with Windows it is maybe not an issue with the board itself or the firmware but a corrupted Linux file.  If that’s true I’m encouraged to keep trying.  However, I only have Macs with OS X El Capitan which prohibits using the USB com port because of a security issue.  That’s why I switched to a Raspberry Pi/Linux environment.  Thanks to you I will keep trying and not return the board for a while.

jwalters1955, 2020-01-27:
Unfortunately I've given up with 1.8.10 on the pi and reverted to 1.8.8. Even after I managed to get .10 to install it threw up Java errors when I tried to run it. Not a Mac person so I can't comment on that aspect but it's damned annoying that .10 won't run on the Pi. I've raised a few comments about it but no real solutions are forthcomming. I'm really sorry I can't help.

leonbrag, 2020-06-05:
The same problem. I had Wifi Rev 2 Uno working perfectly with WiFiNina latest firmware and libary.
I had to upload SSL cert. Upload completed fine, but now I get error:   Communication with WiFi module failed!

djmastrianni, 2020-06-05:
I eventually found the exact line of code that was throwing the error code and determined it was the Linux version of WiFiNinna.  I downloaded the OSX version of the Arduino to an old Mac Mini and ran the Updater and the firmware update worked fine.  I subsequently did the same thing to a second Arduino Uno Rev 2 and that worked fine using the Mac Mini, too.  That was several months ago and both boards are still working correctly on OSX and on Linux - it’s just the Updater that has the bug.  I read a post from someone else in this thread who got it to work with the Windows Version, too.

djmastrianni, 2020-06-05:
To clarify my previous comment to you, the Linux installation I use is on a Raspberry Pi and it may not be the case for other devices running the Arduino IDE Linux version.

leonbrag, 2020-06-05:
Got it to work on windows. Had to power down board, restart dev environment. It worked after that.