arduino/Arduino #9470: Bug: Arduino Nano IOT 33 - Ports & Board Manager

JGPorteous, 2019-11-25:
When double pressing the Arduino Nano IOT 33 to put it in upload mode, the Arduino IDE doesn't handle the change as expected. The Ports Menu shows multiple entries.

Also, after time, the Board Manager also pops up to say "Install this package to use your Arudino NANO IoT board" although, if you click on it, it is already installed.

I've attached a screenshot for your reference

[image]

JGPorteous, 2019-11-25:
If someone can direct me to where the source for this is, I will attempt a fix

per1234, 2019-11-26:
The duplicate ports entries was already reported in https://github.com/arduino/Arduino/issues/8851 and has been fixed in the beta build of the IDE. As for the issue of the spurious package installation notification, I remember also getting this a couple times also related to https://github.com/arduino/Arduino/issues/8851, but not being able to figure out the necessary conditions to reproduce it. I'm guessing that is should also be fixed in the beta build. So I'm going to go ahead and close this, but if you can still reproduce the issue of the spurious package installation notification using the latest beta build then please comment here and I'll re-open the issue.

closed 2019-11-26 by per1234

JGPorteous, 2019-11-26:
Thank you. I will download and test beta