arduino/Arduino #9475: 1.9.0-beta does not compile code with multiple tab’s

mrWheel, 2019-11-27:
All my sketches use multiple tab’s (multiple .ino files) to separate specific functions and make the code readable.
This works great with pré 1.9.0 IDE. All the .ino files are concatenated before send to the compiler. But not 1.9.0-beta.
What am I missing?
Will this be corrected in future releases of 1.9?

matthijskooijman, 2019-11-27:
Can you provide some more details? Such as a minimal example sketch and the compiler errors you get?

mrWheel, 2019-11-27:
Compiling for a Arduino UNO seems to be fine. No errors (had not test that before because I manly use ESP8266 chips..).

But for the ESP8266 all functions declared in an other tab than the main program.ino is "undeclared".

Main program Tab:
`
/*--------------------------------*/
//   FILE: oneWireSearch.ino
//  AUTHOR: Rob Tillaart

#include <OneWire.h>

void setup()
{
  Serial.begin(115200);
  Serial.println("//\n// Start oneWireSearch.ino \n//");

  for (uint8_t pin = 2; pin < 13; pin++)
  {
    findDevices(pin);
  }
  Serial.println("\n//\n// End oneWireSearch.ino \n//");
}

void loop()
{
}
`

The other Tab:
`
/*--------------------------------*/
uint8_t findDevices(int pin)
{
  OneWire ow(pin);

  uint8_t address[8];
  uint8_t count = 0;

  if (ow.search(address))
  {
    Serial.print("\nuint8_t pin");
    Serial.print(pin, DEC);
    Serial.println("[][8] = {");
    do {
      count++;
      Serial.println("  {");
      for (uint8_t i = 0; i < 8; i++)
      {
        Serial.print("0x");
        if (address[i] < 0x10) Serial.print("0");
        Serial.print(address[i], HEX);
        if (i < 7) Serial.print(", ");
      }
      Serial.println("  },");
    } while (ow.search(address));

    Serial.println("};");
    Serial.print("// nr devices found: ");
    Serial.println(count);
  }

  return count;
}
`

Gives this compile error:
`/Users/WillemA/Documents/ArduinoProjects/IDE1.9.0_Test/oneWireSearch/oneWireSearch.ino: In function 'void setup()':
oneWireSearch:26:20: error: 'findDevices' was not declared in this scope
     findDevices(pin);
                    ^
`
**It seems to be related with the ESP8266 2.5 core**

matthijskooijman, 2019-11-27:
Maybe the problem is that it fails to generate a forward declaration for your function. Some additional questions:

 - Can you look for the pre-processed file that is actually being compiled and paste it here? It will be inside some temp directory (which you can figure out by looking at the verbose compilation output). Inside the temp directory, it is `sketch/OneWireSearch.ino.cpp`.
 - If you put the contents of the two files in one file (first the "main program" contents, then the secondary file with your function definition), does it work? If not, then it's not the multiple files that is the problem, but the generation of the forward declaration.
 - You say this works in 1.8.x. Are you sure you're using the same ESP core there?

facchinm, 2019-11-27:
Hi @mrWheel ,
beta IDE uses `arduino-preprocessor` to generate the prototypes; while more powerful than `ctags`, it is based on `clangd` backend so it can't compile some very gcc-specific code that might be contained in the core.
If it errors no prototype is generated and you get the `not declared in this scope` errors

matthijskooijman, 2019-11-27:
> beta IDE uses arduino-preprocessor to generate the prototypes; while more powerful than ctags, it is based on clangd backend so it can't compile some very gcc-specific code that might be contained in the core.

Hm, but that would not explain why it does work for the Uno. Also, the function in question here does not look funny, so I would not expect any problems because of that?

facchinm, 2019-11-27:
It's not about the function itself but the final compilation unit (with `Arduino.h` include and thus the whole core trying to be parsed to generate the AST).
AVR / ARM cores / toolchains don't contain much gcc-ism (or they are properly guarded) so the preprocessor runs happily

mrWheel, 2019-11-27:
> Maybe the problem is that it fails to generate a forward declaration for your function. Some additional questions:
>
> * Can you look for the pre-processed file that is actually being compiled and [...]

[oneWireSearch.ino.cpp.zip](https://github.com/arduino/Arduino/files/3895869/oneWireSearch.ino.cpp.zip)

> * If you put the contents of the two files in one file (first the "main program" contents, then the secondary file with your function definition), does it work? If not, then it's not the multiple files that is the problem, but the generation of the forward declaration.

Well, it is one of the standard examples in the "File->Examples" menu of the IDE that has the `findDevices(int pin)` function in the one-and-only tab.

> * You say this works in 1.8.x. Are you sure you're using the same ESP core there?

Yes, it's the same ESP Core 2.5.2

mrWheel, 2019-11-27:
Hi @facchinm 
Thanks for the explanation .. I don't understand a word of it .. I'm just a user ;-).

matthijskooijman, 2019-11-27:
Interesting, the file you submitted contains the full preprocessed source, rather than just the original source with declarations inserted. I guess this also changed in the beta with arduino-preprocessor?

In any case, the file confirms that the prototype is indeed not being generated. The explanation @facchinm makes sense here: Likely the ESP core contains some gcc-specific constructs that work fine when compiling (which uses the gcc compiler), but break when generating forward declarations (with arduino-preprocessor, which uses a different compiler called clang).

This is probably a bit of a fundamental problem with using clang in this way: It can always produce different results from the gcc compilation, and is more likely to surface in cores with a lot of third party code (such as the ESP cores with the ESP SDK, or the STM32 core with the STM32 HAL).

As a workaround, you can insert the forward declarations explicitly (e.g. put `uint8_t findDevices(int pin);` above your `setup()` function).

mrWheel, 2019-11-27:
@matthijskooijman 
Thanks for figuring this out.
The problem is that I'm not really interested in the `oneWireSearch` example but it infects all of my sketches. Some with a lot of tab's and dozens of functions.... 

Is this something that should be addressed by the ESP8266 core maintainers or is this something you guys will fix? 
The ESP8266 core 2.6x has the same problem but it also breaks all my sketches in a total different way so I don't want to upgrade ...

matthijskooijman, 2019-11-27:
> Is this something that should be addressed by the ESP8266 core maintainers or is this something you guys will fix?

I'm not sure. The ESP core maintainers could maybe fix this by making changes to their code, but if that needs changes to the ESP SDK, that will get tricky with updates. Also, the ESP core is intended to be compiled with gcc, so it might not be reasonable to expect them to fix this.

However, I'm not so sure what can be done at the Arduino end to fix this. Maybe clang could be configured in a more forgiving way?

A guess a good first step would be to figure out what the actual things are that throw off clang. @facchinm is there any way to let arduino-preprocessor output the clang error messages? That sounds like a good first step?