arduino/Arduino #9479: Serial Monitor needs a "Export output" option

alexceltare2, 2019-11-29:
I've lost countless data logged in the Serial Monitor because the blasted Ctrl+A Ctrl+C doesn't function properly or I've accidentally unplugged the serial-based logging device/Arduino. So a good addition to the serial monitor would be a button to export the output as Excel, txt, or any other readable format.

CloudFenrir, 2019-12-16:
Hello, I'll work on this feature  :smile:

AsherThomasBabu, 2020-04-06:
Hey there developers,
I am fairly new to the source code of the Arduino IDE. I happened to want this feature myself and started working on it.

I pulled the latest version from git and `ant built` it and `ant run` the build folder as per instructions from the Wiki page. 

Building and running it the **first time** works flawlessly, But, when built for a **second time** with no changes to the files, the run fails with this message.

```
BUILD FAILED
/home/asher/Arduino/build/build.xml:140: The following error occurred while executing this line:
/home/asher/Arduino/build/build.xml:144: The following error occurred while executing this line:
/home/asher/Arduino/build/build.xml:829: exec returned: 255

Total time: 10 seconds
```

I am running it on a Linux machine, Looking forward to some help.
Thank You 
Asher

per1234, 2021-04-03:
An alternative proposal of making copying Serial Monitor contents to the clipboard is presented at https://github.com/arduino/Arduino/issues/6906

kvkpraneeth, 2021-04-09:
> Hey there developers,
> I am fairly new to the source code of the Arduino IDE. I happened to want this feature myself and started working on it.
>
> I pulled the latest version from git and `ant built` [...]

Hey! did you find a fix for this?

per1234, 2022-12-01:
Now tracked at https://github.com/arduino/arduino-ide/issues/549

closed 2022-12-01 by per1234