arduino/Arduino #9486: Release version not updated in library manager in Arduino IDE

pandayswarnam, 2019-12-02:
I have released a new version of the library in GitHub as version 1.0.7 but the library version in the Library manager is not getting updated, it's still showing the earlier version. What could be the possible reasons for this issue?
Things already tried-
- Updated the version in library properties.
- The release is done properly.

facchinm, 2019-12-02:
@pandayswarnam which library are you referring to? It's a bit difficult to solve a problem without knowing where it is...

pandayswarnam, 2019-12-02:
@facchinm https://github.com/Infineon/DPS310-Pressure-Sensor this is the link to the library.
The main branch is dps310. 
thanks for your quick response.

per1234, 2019-12-02:
@pandayswarnam the reason for this is that the `name` value in library.properties was changed from `DPS310` to `DigitalPressureSensor` after the `V1.0.6` tag was created:
https://github.com/Infineon/DPS310-Pressure-Sensor/commit/54bb28cfe2a139c176c788d11e9aeb384129214a

The Library Manager indexer doesn't support automatically changing the library name, but we can do that for you manually. So let us know if you want the library name to be changed from `DPS310` to `DigitalPressureSensor`.

pandayswarnam, 2019-12-02:
@per1234  yes, please do that manually for me. It would be of great help. Some previous owners did the changes.

Thanks in advance

cmaglie, 2019-12-17:
Done.

closed 2019-12-17 by cmaglie