arduino/Arduino #9493: Playground Code/VisualMicro Page Information Incorrect

psytribalogical, 2019-12-03:
In the below article:
https://playground.arduino.cc/Code/VisualMicro/

It is listed with the content "Continuous free updates for steady compatibility with newest Arduino IDE versions".

This should read "Continuous updates for steady compatibility with newest Arduino IDE versions for licensed users"

VisualMicro, 2019-12-03:
All the updates are available for free in the Microsoft and Microchip Galleries and always have been. Support for older IDE's does end after a few years. For example we are currently supporting Atmel Studio 7, Visual Studio 2017/2019. There is no difference in the availability of updates for anyone.

When moving licences to new machines or after formatting the pc the activation window will appear where you can start one of x free 30 days or activate using a purchased key. The window also asks that people support Visual Micro and buy the annual support but the updates are not currently limited if you do not.

After a significant time from purchase, if people email us requesting licence transfer or reset due to the reasons mentioned above we ask that they buy one years support which is currently $19 for non-commercial use. Again, the updates do not stop if you do not buy the support.

per1234, 2019-12-06:
Hi @VisualMicro. Thanks for your input. It sounds like I can close this issue.

I also appreciate your interest in keeping that Playground page up to date. The Arduino Playground [was made read-only last year](https://forum.arduino.cc/index.php?topic=580991.0). So it is no longer possible for you to directly edit the page. The new procedure is to open an issue here with a description of the changes you want made. An Arduino employee with edit access to the Playground will eventually implement the requested changes.

closed 2019-12-06 by per1234