arduino/Arduino #9499: [Library Manager] TinyCircuits Wireling library

TinyCircuits, 2019-12-05:
Please add the Wireling library found here: https://github.com/TinyCircuits/TinyCircuits-Wireling-Lib

If you have any comments/concerns you can email info@tinycircuits.com, or directly lwienclaw@tinycircuits.com

Thank you!

per1234, 2019-12-05:
Hi @TinyCircuits. You need to create a [release](https://help.github.com/articles/creating-releases/) or [tag](https://git-scm.com/docs/git-tag) that matches the `version` value in your library.properties file.

After doing so, please comment here to let us know we can proceed with adding your library to the Library Manager index.

For more information see:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

TinyCircuits, 2019-12-05:
Accidentally put a v in the tag for just 0.1.0. I deleted that tag and added the correct one without a v!

Thank you!

per1234, 2019-12-05:
Sorry if I wasn't clear. The problem isn't about a tag not exactly matching the `version` value (the Library Manager indexer doesn't care what the tag name is because it uses the `version` value only). The problem is that there are no tags at all in the repository:
https://github.com/TinyCircuits/TinyCircuits-Wireling-Lib/tags

Maybe you created the tag in your local clone and didn't push it to the remote on GitHub?

TinyCircuits, 2019-12-05:
Do you see it now?

per1234, 2019-12-05:
Yes. Everything looks good now. Thanks!

TinyCircuits, 2019-12-05:
WOOH! Thank you!

closed 2019-12-17 by cmaglie