arduino/Arduino #9503: [Library Manager] Add library to library manager

henrygab, 2019-12-06:
Arduino team,

Please add my latest library, QDEC, to the library manager.

Here's the github repo link:

https://github.com/SimpleHacks/EzDmaHelper

Thank you

per1234, 2019-12-06:
Hi @henrygab. You need to create a [release](https://help.github.com/articles/creating-releases/) or [tag](https://git-scm.com/docs/git-tag) that matches the `version` value in your library.properties file.

After doing so, please comment here to let us know we can proceed with adding your library to the Library Manager index.

For more information see:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

henrygab, 2019-12-07:
@per1234 -- Oops!  Forgot to add the tag.  Updated now, thanks for the reminder.

cmaglie, 2019-12-17:
Done, sorry for the long delay

closed 2019-12-17 by cmaglie

njh, 2020-07-04:
Looks like this repo has now been deleted?
Any information about why?

henrygab, 2020-07-08:
Three reasons:
1. It was a work in progress
2. That work never finished (substantial bugs discovered)
3. No ongoing support

Therefore, its continued existence would just cause pain, not be helpful.

It's not fully gone, just private now.  If the time/resources to fix and fully test the library arrive, it will be revived.