arduino/Arduino #9504: [Library Manager] Please add Adafruit ICM20649

siddacious, 2019-12-07:
Please add our new library to the library manager:
https://github.com/adafruit/Adafruit_ICM20649

Thanks!

per1234, 2019-12-07:
Hi @siddacious. 
You need to create a [release](https://help.github.com/articles/creating-releases/) or [tag](https://git-scm.com/docs/git-tag) that matches the `version` value in your library.properties file.

After doing so, please comment here to let us know we can proceed with adding your library to the Library Manager index.

For more information see:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

siddacious, 2019-12-07:
Whoopsies! Done, must have been in a rush
https://github.com/adafruit/Adafruit_ICM20649/releases/tag/1.0.0

cmaglie, 2019-12-17:
Done, sorry for the long delay

closed 2019-12-17 by cmaglie

siddacious, 2019-12-17:
@cmaglie no worries, thanks for you help!