arduino/Arduino #9505: Sketch can not be compiled after removing a file.

tikonen, 2019-12-07:
Arduino IDE 1.8.10 On Windows 10.

Steps to reproduce.

1. Open sketch and verify/compile.
2. Add new file to project (Sketch/Add file..)
3. Verify/compile the sketch
4. Delete the file you just added from the IDE menu
5. Verify/compile the sketch

What happens:
Compile fails with an error 
```
unable to create a folder to save the sketch: mkdir sketch: Access is denied. 
Error compiling for board Arduino Nano.
```

With verbose mode the error is:
```
C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\Teemu Ikonen\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\Teemu Ikonen\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Teemu Ikonen\Documents\Arduino\libraries -fqbn=arduino:avr:nano:cpu=atmega328old -ide-version=10810 -build-path C:\Users\TEEMUI~1\AppData\Local\Temp\arduino_build_791852 -warnings=default -build-cache C:\Users\TEEMUI~1\AppData\Local\Temp\arduino_cache_564665 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avrdude.path=C:\Users\Teemu Ikonen\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17 -prefs=runtime.tools.avrdude-6.3.0-arduino17.path=C:\Users\Teemu Ikonen\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17 -prefs=runtime.tools.avr-gcc.path=C:\Users\Teemu Ikonen\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino5 -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino5.path=C:\Users\Teemu Ikonen\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino5 -prefs=runtime.tools.arduinoOTA.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.arduinoOTA-1.3.0.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -verbose C:\Users\Teemu Ikonen\Documents\Arduino\geiger_testbench\geiger_testbench.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\Teemu Ikonen\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\Teemu Ikonen\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Teemu Ikonen\Documents\Arduino\libraries -fqbn=arduino:avr:nano:cpu=atmega328old -ide-version=10810 -build-path C:\Users\TEEMUI~1\AppData\Local\Temp\arduino_build_791852 -warnings=default -build-cache C:\Users\TEEMUI~1\AppData\Local\Temp\arduino_cache_564665 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avrdude.path=C:\Users\Teemu Ikonen\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17 -prefs=runtime.tools.avrdude-6.3.0-arduino17.path=C:\Users\Teemu Ikonen\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17 -prefs=runtime.tools.avr-gcc.path=C:\Users\Teemu Ikonen\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino5 -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino5.path=C:\Users\Teemu Ikonen\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino5 -prefs=runtime.tools.arduinoOTA.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.arduinoOTA-1.3.0.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -verbose C:\Users\Teemu Ikonen\Documents\Arduino\geiger_testbench\geiger_testbench.ino
Using board 'nano' from platform in folder: C:\Program
Using core 'arduino' from platform in folder: C:\Program
unable to create a folder to save the sketch: mkdir C:\Users\TEEMUI~1\AppData\Local\Temp\arduino_build_791852\sketch: Access is denied.
Error compiling for board Arduino Nano.
```

per1234, 2019-12-07:
Hi @tikonen. I have not been able to reproduce this bug. It's possible it could be caused by your antivirus software. Try **TEMPORARILY** disabling the antivirus for a single compilation, then turn the antivirus back on. Please let us know whether the error still occurs with the antivirus disabled.

tikonen, 2019-12-07:
Hi, I don't have antivirus so this is probably caused by something else. I checked the build folder when error happens and there is a sketch folder that can not be accessed.

```
c:\Users\Teemu Ikonen\AppData\Local\Temp\arduino_build_277221>dir
 Volume in drive C has no label.
 Volume Serial Number is BC4F-7093

 Directory of c:\Users\Teemu Ikonen\AppData\Local\Temp\arduino_build_277221

12/07/2019  06:57 PM    <DIR>          .
12/07/2019  06:57 PM    <DIR>          ..
12/07/2019  06:57 PM             1,507 build.options.json
12/07/2019  06:57 PM    <DIR>          sketch
               1 File(s)          1,507 bytes
               3 Dir(s)  78,356,312,064 bytes free

c:\Users\Teemu Ikonen\AppData\Local\Temp\arduino_build_277221>cd sketch
Access is denied.

c:\Users\Teemu Ikonen\AppData\Local\Temp\arduino_build_277221>
```
After I close the Arduino IDE the both the json and folder sketch disappear from the build folder.

np703, 2020-01-17:
I just had exactly the same problem, trying to use tabs to organise a complex Arduino application.

SpamWall, 2020-05-15:
The problem is that the folder created by the IDE (c:\Users\Teemu Ikonen\AppData\Local\Temp\arduino_build_277221 in @tikonen 's post above is created without an owner.  Windows does not give you permission to delete an unowned forler (even if you are the administrator).

michele-bertoni, 2020-05-25:
I had the very same issue on Windows after removing a .h file from the IDE. Fixed just by closing and reopening Arduino IDE

pixelk0, 2020-06-09:
The bug is still in 1.8.12 on June 2020.

mcuprojects, 2020-07-20:
On 1.9.0 beta still present. (on windows 10)

For those who might have come here from google in search for solution:
The way to get around this issue is to close the Arduino IDE and re-open your sketch. Then it will work fine.

Kos-M, 2020-09-28:
The bug is still in 1.8.13

per1234, 2021-04-06:
It looks like there was a previous report at https://github.com/arduino/Arduino/issues/7229

Chaos99, 2021-10-30:
Still in 1.8.16. Re-starting helped as described.

Andy2No, 2024-07-26:
Bug still present in 1.8.19.