arduino/Arduino #9507: Libraries in sketch folder still not working

jeroenst, 2019-12-07:
According to the manual libraries can be placed in a libraries folder in the sketch folder.

https://www.arduino.cc/en/Guide/Libraries

> Arduino libraries are managed in three different places: inside the IDE installation folder, inside the core folder and in the libraries folder inside your sketchbook. 

But it doesn't work

Related bug: https://github.com/arduino/Arduino/issues/4936

matthijskooijman, 2019-12-07:
I assume you refer to the "inside your sketchbook" part? The sketchbook is the directory where your sketches live (e.g. `Documents/Arduino` or something), and libraries can live in there (e.g. `Docuements/Arduino/libraries/my_library`). AFAIU the docs are not trying to say "libraries can be loaded from your **sketch** folder", so I think the docs are correct and this feature is (unfortunately) still missing.

I'm going to close this issue as invalid. If you feel this is not appropriate, let use know through a comment. However note #7755, which is an issue that requests this specific feature and should be used for any further discussion about the feature (so no need to keep this open as a feature request).

closed 2019-12-07 by matthijskooijman

jeroenst, 2019-12-07:
The statement says literally libraries folder inside your sketchbook which imo is the project folder, not the projectS folder because it's not stated sketchbookS folder.

This feature is also very wanted, because for different projects sometimes the same lib is needed but with different version. That's now almost impossible.

Also when a project on github has submodules of libraries included they now have to be copied out of the sketchbook folder to the libraries folder in the folder containing all the sketchbooks.

But still even if the only location is in the sketchbook folder we also want to have global libraries locating in folder containing all the sketchbooks (like the behaviour is now).

matthijskooijman, 2019-12-07:
> The statement says literally libraries folder inside your sketchbook which imo is the project folder, not the projectS folder because it's not stated sketchbookS folder.

This is a misunderstanding on your side. Check for example the preferences of the Arduino IDE, which lets you set the *sketchbook* directory. The *sketchbook* contains a number of *sketches*, which is what you are calling projects.

One can argue about how intuitive or clearly documented this is, but I am pretty sure this is how the terminology was intended.

As for the rest of your argument: Agreed, but that is for #7755.