arduino/Arduino #9516: Warning message on system log when IDE is showing a "matching brace" popup

jyganci, 2019-12-11:
I'm using Arduino 1.8.10 on fedora 30 linux. While working on a sketch, I happened to have the fedora system log displaying in a terminal window so that I could monitor the status of the serial port being used by the Arduino IDE. When I cllick my mouse just to the right of the ending brace "}" for the loop() code, a small popup appeared at the top of my sketch showing

    void loop()
    {

since not all of my loop() code fit the window. This is nice -- I see the where the "matching brace" for the one where my mouse is positioned. However, the following messages appear on the system log whenever I move the mouse to the ending brace and the matching beginning brace is not visible:

Dec 11 09:53:07 xxxxxxxxxxxxxxxxi org.gnome.Shell.desktop[1952]: Window manager warning: Window 0x260020f sets an MWM hint indicating it isn't resizable, but sets min size 1 x 1 and max size 2147483647 x 2147483647; this doesn't make much sense.

Note: for my own security, the "xxxxxxxxxxxxxxx.org" replaces what is really shown on my PC.

This does not cause a problem; it's just clutters the system log. Just pointing it out to you. Would be nice to eliminate the warning message bui I can certainly live with it since I do not normally view the system log while working on a sketch.

I did pose this to the Arduino forum, where it was recommended that I verify that the issue still exists with the latest beta and hourly build, then post a bug report if the issue still exists. The issue still exists in both the latest beta (1.9.0) and hourly build (1.8.11 Hourly Build 2018/12/09 01.25).

Steps to reproduce.
1. Open terminal window.
2. Issue "journalctl -f" or equivalent (before systemd I entered "tail /var/log/messages -f").
3. Start the Arduino IDE.
4. Open a new sketch: click File->New.
5. New sketch appears. Click within the loop() code area. Press Enter repeatedly until the "loop() {
scrolls off the top of the window.
6. Use the down arrow key to scroll down until the ending brace for loop() appears.
7.Click your mouse to the right of the ending brace.
8. "Matching brace" popup appears at the top of the sketch.
9. Message described above appears in the terminal window showing the system log.

Regards, John Ganci

ubdussamad, 2020-03-27:
Hi @jyganci  , what window manager are you using? Metacity? , this could just be a issue with your window manager as the IDE could not have been tested with your specific window manager. It's working fine in compiz which ships with ubuntu. You can check your window manager using [this](https://askubuntu.com/questions/72549/how-to-determine-which-window-manager-is-running) resource. My advise , if it bug you too much try switching the window manager you currently have, you can change it back  to default as well.

jyganci, 2020-06-29:
Sorry for the late reply. I haven't checked this email account in a while.

I run fedora linux. It uses the wayland display manager. When I log in to
my session, I am using Gnome.

Regards,

John Ganci