arduino/Arduino #9520: Incorrect permissions on libraries in nightly releases causing problems on multi-user setups

maxnet, 2019-12-12:
Lately some of the .jar files inside https://downloads.arduino.cc/arduino-nightly-linuxarm.tar.xz have incorrect file permissions.

It concerns the following files which are mode 0600:

java/lib

```
-rw------- 1 max max  3135610 Jul 22 20:45 charsets.jar
-rw------- 1 max max    96657 Jul 22 20:45 jce.jar
-rw------- 1 max max   684994 Jul 22 20:45 jsse.jar
```

java/lib/ext

```
-rw------- 1 max max 3860563 Jul 22 20:45 cldrdata.jar
-rw------- 1 max max    8286 Jul 22 20:45 dnsns.jar
-rw------- 1 max max   44516 Jul 22 20:45 jaccess.jar
-rw------- 1 max max 1179126 Jul 22 20:45 localedata.jar
-rw------- 1 max max 2033511 Jul 22 20:45 nashorn.jar
-rw------- 1 max max   56313 Jul 22 20:45 sunec.jar
-rw------- 1 max max  265907 Jul 22 20:45 sunjce_provider.jar
-rw------- 1 max max  244507 Jul 22 20:45 sunpkcs11.jar
-rw------- 1 max max   68974 Jul 22 20:45 zipfs.jar
```

java/lib/security/policy/limited

```
-rw------- 1 max max 1040 Jul 22 20:45 local_policy.jar
-rw------- 1 max max  620 Jul 22 20:45 US_export_policy.jar
```

java/lib/security/policy/unlimited

```
-rw------- 1 max max 638 Jul 22 20:45 local_policy.jar
-rw------- 1 max max 620 Jul 22 20:45 US_export_policy.jar
```

They should be 0644 -so other users can read them- or Arduino will fail to start on multi-user setups in which Arduino is installed in a global directory by the root user, but run by different users.

facchinm, 2019-12-16:
Will fix on the Jenkins side, thanks!

facchinm, 2019-12-16:
Should be fixed now, reopen if needed. Thanks for the feedback!

closed 2019-12-16 by facchinm