arduino/Arduino #9523: [Feature Request] Add SoftwareSerial / Uart lib to Nano 33

pierre-b, 2019-12-14:
Hello Arduino,

For an IoT project I need to connect a GPS and GPRS board, each needs a Serial port.

As there is only one Serial port available (the USB is taken) I need a second Serial port. Unfortunately SoftwareSerial and Uart libs don't work on Nano 33 and I'm stuck.

Is it something you could add? or clearly I can't build my project with a Nano 33?

Thanks

per1234, 2019-12-14:
Which Nano 33 are you using, the Nano 33 IoT or the Nano 33 BLE/BLE Sense?

pierre-b, 2019-12-14:
It's BLE Sense

facchinm, 2019-12-16:
Hi @pierre-b ,
Nano 33 BLE has a second serial port (as `Serial1`) on pins 0 and 1. You can use it to communicate with the GPS. If the library you are using is not compiling please add some info here (like the hardware you are targeting and the library name and version)