arduino/Arduino #9526: [Library Manager] ArduinoSIM800L

carrascoacd, 2019-12-15:
Hello.

I created a HTTP and FTP client library for SIM800 and Arduino.
Could you add this to a list of libraries?
url: https://github.com/carrascoacd/ArduinoSIM800L

Thanks!

per1234, 2019-12-15:
Hi @carrascoacd 
Hi. Arduino recently started a new policy that 3rd party libraries with a library.properties `name` value starting with the word "Arduino" (or any capitalization thereof) will no longer be accepted for inclusion in the Arduino Library manager index. The reason is that Arduino wants to reserve these names for their own official libraries.

If you want your library to be added to the Library Manager index, you will need to change the `name` value in library properties to one that is compliant with the Arduino Library Specification:
https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5:-Library-specification#libraryproperties-file-format

Then update the `version` value in library.properties and create a new release/tag, or else redo the existing release/tag.

Note that this restriction only applies to the `name` value. You are not required to rename the repository or header files.

Once you have done this, please comment here so we'll know we can proceed.

More information:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

carrascoacd, 2019-12-15:
Hi! Thanks for your response.

And, what should I do to make it official?
Otherwise I can change the name. Will it be searcheable or do you index by
name only?

Regards,

per1234, 2019-12-15:
>what should I do to make it official?

By "official", I mean the libraries that Arduino maintains, most of which are listed here:
https://github.com/arduino-libraries

Since Arduino does not make any hardware that uses the SIM800L, it's very unlikely they would want to take over the maintenance of your library.

>Otherwise I can change the name.

If you want your library in the Library Manager index, that's the best solution.

>Will it be searcheable or do you index by
name only?

The Library Manager's search scope is the `name`, `sentence`, `paragraph`, and `includes` fields of library.properties.

cmaglie, 2020-02-17:
I'm closing this for now, please open a new request when the library is ready.

closed 2020-02-17 by cmaglie