arduino/Arduino #9530: Please update documentation for Arduino UNO WiFi Rev2

PickyBiker, 2019-12-17:
This is a followup to an arduino forum post at https://forum.arduino.cc/index.php?topic=654029.0 and another at https://forum.arduino.cc/index.php?topic=625019.0

People familiar with the Arduino UNO are not aware that the SDA and SCL pins are not located at A4 and A5 on UNO WiFi. They are located on the two unmarked pins near AREF.  Please update the following document to include UNO WiFi Rev2 information.
https://www.arduino.cc/en/Reference/Wire

Also, please create a definitive UNO WiFi Rev2 pin-out document.

facchinm, 2019-12-17:
Hi @PickyBiker ,
on UNO WiFi rev2 the pins are clearly marked as SDA and SCL on the header (and yes, they are near AREF as per UNO r3 specifications). I think @per1234 answer on the first post can be used as reference to improve the website documentation.

PickyBiker, 2019-12-17:
Yes, you are correct. I was just looking at the markings on the board itself and didn't look at the header. My bad.

dushyant18033, 2020-03-12:
Hi,
I've seen Uno R3 also has these 2 pins marked as SCL and SDA for I2C near the AREF pin header. But since Uno has these also connected to A5 and A4, people don't usually notice these.

FunzioneFabio, 2020-06-30:
Hi @PickyBiker,
Pinout documentation is now online in the documentation section on product pages for both UNO rev3 and UNO wifi rev2.
https://content.arduino.cc/assets/Pinout-UNOrev3_latest.pdf
https://content.arduino.cc/assets/Pinout-UNOwifirev2_latest.pdf

per1234, 2020-06-30:
Thanks for the update @FunzioneFabio!

Are you able to update the I2C pin table on the Wire library reference's home page to add the Uno WiFi Rev2 to the list (and hopefully the other missing official boards while you're at it)?

It would also be nice to have this documented in the [Uno WiFi Rev2's "Getting Started" page](https://www.arduino.cc/en/Guide/ArduinoUnoWiFiRev2#toc8) (which does already list some of the other differences from the classic Uno).

If we could get that updated, then we would be able to close this issue report as solved.

FunzioneFabio, 2020-06-30:
Sorry @per1234 but I'm not able to do that.

per1234, 2020-10-01:
>It would also be nice to have this documented in the Uno WiFi Rev2's "Getting Started" page (which does already list some of the other differences from the classic Uno).

This has now been done:
https://www.arduino.cc/en/Guide/ArduinoUnoWiFiRev2#i2c

The requirement to add the information to the [Wire library reference page](https://www.arduino.cc/en/Reference/Wire) remains, so I leave this issue open.