arduino/Arduino #9532: [Library Manager] ESP_WiFiManager for ESP32 and ESP8266

khoih-prog, 2019-12-17:
https://github.com/khoih-prog/ESP_WiFiManager

This library is based on, modified, bug-fixed and improved from:

[Tzapu WiFiManager] (https://github.com/tzapu/WiFiManager)
[Ken Taylor WiFiManager] (https://github.com/kentaylor/WiFiManager)
to add support to ESP32 besides ESP8266.

This is an ESP32 / ESP8266 WiFi Connection manager with fallback web configuration portal. It's using a web configuration portal, served from the ESP32 / ESP8266, and operating as an access point.

The configuration portal is captive, so it will present the configuration dialogue regardless of the web address selected, excluding https requests.

Thanks,

cmaglie, 2019-12-17:
Done

closed 2019-12-17 by cmaglie

khoih-prog, 2019-12-18:
Thanks so much. and Best Regards.