arduino/Arduino #9535: SD.exists(filename) faulty

dsyleixa, 2019-12-18:
SD.exists(filename) works faulty!
If the SD has been started correctly by SD.begin(SD_CS) and the SD should be checked if intermediately removed, it always shows "OK" and never throws an error:

```
      if( SD.exists("/") ) {
            Serial.println((String)" SD path / OK!  ");  // either if SD inserted or removed !!!
            // debug
            delay(2000);
            readDirectory(SdPath, 0);
         }
         else {
            Serial.println((String)"no SD file path found!  "); 
            // debug
            delay(2000);
         }

```

dsyleixa, 2019-12-23:
any hints...?

matthijskooijman, 2019-12-23:
I suspect that `SD.exists("/")` always returns true immediately, even without looking at the SD card, so maybe this is intended behaviour. Did you look at the code inside the SD library to see what happens?

Is there perhaps some method you can call that detects whether the SD card is (still) present? Or is that only the return value of `SD.begin()`?

dsyleixa, 2019-12-23:
the root path is needed to check 
SD.exists("/")
because the current  files on the SD are unknown, So the root pat is the only path only which is existing on either formatted SD.

I hve no idea what the lib does or not, but if intended behavior o not: it's faulty how it works currentl, so it needs to be fixed either way.

dsyleixa, 2019-12-31:
when wil there finally be a "bug" tag and probably a fix?

per1234, 2019-12-31:
@dsyleixa you're reporting this in the wrong place. The 3rd party ESP32 hardware core comes with its own version of the SD library:
https://github.com/espressif/arduino-esp32/tree/master/libraries/SD
In fact, the official Arduino SD library works exactly the opposite of what you describe. It will always trigger the `else` statement of your code.

If you were to want to to make a feature request or report bugs in the official Arduino SD library, then this is still the wrong place for it, since this repository doesn't contain that code. The official Arduino SD library repository is here:
https://github.com/arduino-libraries/sd/

Please, please, in the future if you're going to submit issue reports, make sure to do it to the right repository, I know I've been over this with you before multiple times in the past. When you go to submit an issue report, you are asked to read the contributing rules, which very clearly told you where to report this issue:
https://github.com/arduino/Arduino/blob/master/CONTRIBUTING.md#issues
>Submit issue reports to the correct repository:
>
>Issue topic | Report at
>-- | --
>3rd party libraries, hardware, or sketches | Report issues to the author of the software, not Arduino.

and:

>Arduino libraries | [arduino-libraries](https://github.com/arduino-libraries)
>-- | --

---
Spamming the issue thread with nagging comments is in no way productive. In fact it is likely to cause your issue to be locked and made less of a priority. 978 people get an email every time you comment on an issue here. That's fine if you have something productive to add to the conversation, but things like "any hints or comments...?" and "when wil there finally be a "bug" tag and probably a fix?" are completely unproductive.

per1234, 2021-03-28:
Closing as off topic for this repository. An equivalent feature request, based on the actual  behavior of Arduino's library and submitted to [the appropriate repository](https://github.com/arduino-libraries/SD), would be welcome.

closed 2021-03-28 by per1234