arduino/Arduino #9539: Ethernet Library Error on IDE 1.8.10?

RichardSelmanXX, 2019-12-18:
Just installed IDE 1.8.10.  Controllino Mega.  On compile get ....

In file included from C:\Program Files (x86)\Arduino\libraries\Ethernet\src\Dns.cpp:8:0:

C:\Program Files (x86)\Arduino\libraries\Ethernet\src\Dns.cpp: In member function 'uint16_t DNSClient::BuildRequest(const char*)':

C:\Program Files (x86)\Arduino\libraries\Ethernet\src\utility/w5100.h:457:25: warning: result of '(256 << 8)' requires 18 bits to represent, but 'int' only has 16 bits [-Wshift-overflow=]

 #define htons(x) ( (((x)<<8)&0xFF00) | (((x)>>8)&0xFF) )

                      ~~~^~~

C:\Program Files (x86)\Arduino\libraries\Ethernet\src\Dns.cpp:164:18: note: in expansion of macro 'htons'

  twoByteBuffer = htons(QUERY_FLAG | OPCODE_STANDARD_QUERY | RECURSION_DESIRED_FLAG);

                  ^~~~~

matthijskooijman, 2019-12-18:
Seems like a warning, which is also harmless (code works as expected AFAICS). 

Still would be good to fix, though. The code for this library lives at https://github.com/arduino-libraries/Ethernet, so maybe you could re-report the bug there? Don't forget to look for an existing issue about this, of course.

per1234, 2019-12-18:
Moved to: https://github.com/arduino-libraries/Ethernet/issues/122

closed 2019-12-18 by per1234