arduino/Arduino #9554: V 1.8.10 Windows 10

cfar1900, 2019-12-24:
Header of IDE still reports version 1.8.9 after update.  Pop-up asks to update to version 1.8.10 then says it's already installed.

per1234, 2019-12-24:
Closing as duplicate of https://github.com/arduino/Arduino/issues/9399

closed 2019-12-24 by per1234

SasmitoMotion, 2019-12-24:
hello sir
iam have troubleshooting in my board
when iam compile ESP32 - with windows 7 and Arduino IDE

Global variables use 53488 bytes (16%) of dynamic memory, leaving 274192
bytes for local variables. Maximum is 327680 bytes.
Sketch too big; see http://www.arduino.cc/en/Guide/Troubleshooting#size for
tips on reducing it.
Error compiling for board ESP32 Dev Module.

per1234, 2019-12-24:
@SasmitoMotion this issue tracker is only to be used to report bugs or feature requests. This topic is more appropriate for the Arduino Forum. I'm sure we'll be able to help you with your problem over there.

Please do this:
1. Read http://forum.arduino.cc/index.php?topic=148850
2. If you haven't already done so, create a new thread in the appropriate section of the forum (http://forum.arduino.cc/), following all the rules. Be sure to post your code, using code tags (`</>` button on the toolbar).

SasmitoMotion, 2019-12-24:
@per1234 okay thank you Mr Per1234 for bring these forum

SasmitoMotion, 2019-12-25:
i have an issue on board ESP32, how to connect wireless camera
that task

it crashes on the send() call with:

Got request
Guru Meditation Error: Core  1 panic'ed (Cache disabled but cached
memory region accessed)
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction).
Exception was unhandled.
Memory dump at 0x40152abc: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction).
Exception was unhandled.
Memory dump at 0x40152abc: bad00bad bad00bad bad00bad
Guru Meditation Error: CoGuru Meditation Error: Core  1 panic'ed (

it doesn't finish the last printout, that's why it's truncated in the
message above, and it hangs there, doesn't even reset the board.
then include IR sensor