arduino/Arduino #9559: one simple function call make code bigger more then 100 bytes

dvanto, 2019-12-26:
one simple call of fuction read_flash() - without params make grow of binnary code on 150 bytes
this function already used in setup()
addtional call in PumpOff (in binery code it spent 8 bytes)
`        	->

        	0e 94 0c 0a 	call	0x1418	; 0x1418 <read_flash()>
        	8f ea       	ldi	r24, 0xAF	; 175
        	93 e0       	ldi	r25, 0x03	; 3
`

I am using arduino pro 168/16

full scetch sources in archive:
[wp.zip](https://github.com/arduino/Arduino/files/4002502/wp.zip)

PS. I understand that this code could by optimized and so on.. but I need answer for one question: why 8-bytes call make binary code grow on 150 bytes

> с read_flash()
Скетч использует 14460 байт (100%) памяти устройства. Всего доступно 14336 байт.
Глобальные переменные используют 802 байт (78%) динамической памяти, оставляя 222 байт для локальных переменных. Максимум: 1024 байт.

> Скетч использует 14306 байт (99%) памяти устройства. Всего доступно 14336 байт.
Глобальные переменные используют 802 байт (78%) динамической памяти, оставляя 222 байт для локальных переменных. Максимум: 1024 байт.

per1234, 2019-12-26:
This issue tracker is only to be used to report bugs or feature requests. This topic is more appropriate for the Arduino Forum. I'm sure we'll be able to help you with your problem over there.

Please do this:
1. Read http://forum.arduino.cc/index.php?topic=148850
2. If you haven't already done so, create a new thread in the appropriate section of the forum (http://forum.arduino.cc/), following all the rules. Be sure to post your code, using code tags (`</>` button on the toolbar).

closed 2019-12-26 by per1234

dvanto, 2019-12-26:
this is bug - small call (8 bytes) make binary code grow on 150 bytes