arduino/Arduino #9560: BUG: one additional simple call make code grow more by 150 bytes

dvanto, 2019-12-26:
one simple call of fuction read_flash() - without params make grow of binnary code on 150 bytes

addtional call in PumpOff (in binary code it spent 8 bytes)
` ->

    	0e 94 0c 0a 	call	0x1418	; 0x1418 <read_flash()>
    	8f ea       	ldi	r24, 0xAF	; 175
    	93 e0       	ldi	r25, 0x03	; 3
`
this function already called in setup()
without additional call size is 14306, with call - 14460 bytes

I am using arduino pro 168/16

full scetch sources in archive:
[wp.zip](https://github.com/arduino/Arduino/files/4002842/wp.zip)

with additional call of read_flash()

> Скетч использует 14460 байт (100%) памяти устройства. Всего доступно 14336 байт.
> Глобальные переменные используют 802 байт (78%) динамической памяти, оставляя 222 байт для локальных переменных. Максимум: 1024 байт.

without call:

> Скетч использует 14306 байт (99%) памяти устройства. Всего доступно 14336 байт.
> Глобальные переменные используют 802 байт (78%) динамической памяти, оставляя 222 байт для локальных переменных. Максимум: 1024 байт.

facchinm, 2019-12-30:
Hi @dvanto ,
unused functions get stripped by the linker (if using LTO, and AVR core does) so calling the function (8 bytes) also brings in the binary the object code of that function (hence the ~150bytes increase)

closed 2019-12-30 by facchinm

dvanto, 2020-02-07:
Hi

My function used 
But second call of same function make code to grow

Although I see that function code in hex same and call needs only 8 bytes

Отправлено из мобильной Почты Mail.ru

понедельник,

cmaglie, 2020-02-07:
What else is added, besides the 8 bytes of the function call?

Since you already examined the disassembled, it should not be difficult to find where the extra 150 bytes are...

dvanto, 2020-02-07:
Nothing. Just one call. 
Hex files has a lot of differences 
I can check only few places in asm: called function’s code and place of call 

Отправлено из мобильной Почты Mail.ru

matthijskooijman, 2020-02-07:
I once made a script that helps to compare machine code between .elf files (removing absolute addresses to reduce noise). In case it is helpful, you can find it here: https://github.com/matthijskooijman/scripts/blob/master/Arduino/diff-asm

cmaglie, 2020-02-07:
> Hex files has a lot of differences 
> I can check only few places in asm: called function’s code and place of call 

hex and asm match 1:1: this means that each byte in the hex must have an equivalent instruction in the asm. Just compare the asm with and without the call to see what changes (besides the function call), you should easily find where those 150 bytes are...

BTW you have to do this if you want: I don't think that someone else wiil do that for you.