arduino/Arduino #9562: Arduino IDE confuses *.h files with *.ino sketches

aphcarios, 2019-12-27:
I'm using arduino IDE version 1.8.10 it was working fine but suddenly with I open the project my headers files show up in the first tap of my project (See attached Image). this makes arduino IDE unable to compile the sketch correctly.
[image]

I'm using vsCode as external editor with arduino plug. external editor is checked in the arduino IDE preferences. the project is also a git repository. the compile verbose shows the following:
`avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10810 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\cores\\arduino" "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\variants\\standard" "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\libraries\\EEPROM\\src" "C:\\Users\\******\\AppData\\Local\\Temp\\arduino_build_869209\\sketch\\HW_config.h.cpp" -o nul
Alternatives for HW_config.h: []HotboxArduino:3:10: error: HW_config.h: No such file or directory`

in my preferences file is says that my last sketch opened if ***.h file instead of header file.
[image]

so far I tried the following:

1- creating new project whit *.ino file and *.h file ( worked) this mean the IDE does not get confused all the time.
2- creating new folder with sketch name (in the Desktop) and copy the projects file to is (Same issue)
3- removing the .git file (Same issue)
4- uninstalling and installing Arduino IDE (Same issue)
5- Manually editing preferences.txt (Same issue)
6- compiling with VSCode (Same issue)

could you please verify if this is a bug in the IDE ?? or if any one has similar issue

aphcarios, 2019-12-27:
renaming the sketch and its containing folder does not work either

magedrifaat, 2020-01-02:
Are you able to replicate this issue in a different sketch?

aphcarios, 2020-01-02:
I copied every thing to new sketch and to new location that worked for my 
I'll try to clone the repository to new computer and see if this issue still comes out

cmaglie, 2020-01-29:
Check that the containing folder name match the main .ino file: if your sketch is named `MySketch.ino` it must be in inside a folder named `MySketch`.

per1234, 2021-05-22:
Thanks for taking the time to submit an issue @aphcarios .

I see we already have a prior issue report about this at https://github.com/arduino/Arduino/issues/8030.

It's best to have only a single issue per subject so we can consolidate all relevant discussion to one place, so I'll go ahead and close this in favor of the other.

If you end up with additional information to share, feel free to comment in the other thread.

closed 2021-05-22 by per1234