arduino/Arduino #9564: Check if the board is supported by more than one of the installed cores.

JazzTp, 2019-12-27:
### Suggestion:

**Maybe the Arduino IDE boards manager could check (or check in some other manner if it already does) if support for the same hardware is being provided by more than one of the installed cores. And again maybe this could be done after selecting the board, and/or before starting to compile anything, and possibly even provide the option to chose among the cores found for the board, or more simply print out their paths and invite to remove (or move away) all but one.**

---

### Details about _why_ I'm posting the suggestion above.

I have at the moment a superficial knowledge of the Arduino IDE inner "mechanisms", apology if this feedback is useless or if it should be reported elsewhere.

I had posted here at first: https://github.com/esp8266/Arduino/issues/6503

**Arduino 1.8.10**

I spent a few hours with **error messages** which where not pointing me to the real problem (even compiling the bare minimum sketch with empty setup() and loop() functions), **both after following the instructions to install the [ESP8266 core](https://github.com/esp8266/Arduino) via Boards Manager _and_ [from git](https://arduino-esp8266.readthedocs.io/en/2.5.0/installing.html#using-git-version)**, until I found [the issue already linked above](https://github.com/esp8266/Arduino/issues/6503) and **realized** that **I had some other core supporting esp8266** (which I had installed earlier, _together with other cores_, for boards with which the IDE is doing fine).

**After moving away the folder hardware/SparkFunBoard_Addons_Jan_16/esp8266/** (actually 7zipping with password to be sure, than `rm -rf`), **the [ESP8266 core](https://github.com/esp8266/Arduino) is working just great with the Arduino IDE** (git install at the moment, just because I was trying out everything including having it under portable/ or not).

per1234, 2021-04-02:
Closing as fixed by https://github.com/arduino/Arduino/pull/9238
This is a super nice advancement made in the Arduino IDE 1.8.13 release that will make it much more clear which boards platform the board you have currently selected is from.

closed 2021-04-02 by per1234

JazzTp, 2021-04-03:
Great, thank you!