arduino/Arduino #9576: A simple and handy data logging library

matusm, 2019-12-31:
This is a simple and lightweight library for all Arduino boards. 
Please consider to add it to the library manager.

https://github.com/matusm/Arduino-DataSeriesPod

SasmitoMotion, 2019-12-31:
Hello Mr can someone help me
iam installing Esp32 Cam in Arduino IDE
iam compile Camera Web Server - Done
when iam upload the code

Arduino: 1.8.10 (Windows 7), Board: "AI Thinker ESP32-CAM"

Sketch uses 2101295 bytes (66%) of program storage space. Maximum is
3145728 bytes.
Global variables use 53576 bytes (16%) of dynamic memory, leaving 274104
bytes for local variables. Maximum is 327680 bytes.
esptool.py v2.6
Serial port COM3
Connecting........_____....._____....._____....._____....._____....._____....._____

A fatal error occurred: Failed to connect to ESP32: Timed out waiting for
packet header
A fatal error occurred: Failed to connect to ESP32: Timed out waiting for
packet header

cmaglie, 2020-01-07:
Done.

closed 2020-01-07 by cmaglie