arduino/Arduino #9578: bad cpu type in executable

sh1va73, 2020-01-01:
On Mac Catalina I get this error: bad cpu type in executable
How to proceed?

per1234, 2020-01-01:
Closing as duplicate of https://github.com/arduino/Arduino/issues/8974

closed 2020-01-01 by per1234

sh1va73, 2020-01-01:
I know this question has been asked before but the solutions is not clear to me. What is the solutions to this?

per1234, 2020-01-01:
@sh1va73 this isn't an appropriate place to request assistance. This is a place to report bugs and submit feature requests. The place to request assistance is the Arduino Forum. I'm sure we'll be able to help you with your problem over there.

Please do this:
1. Read http://forum.arduino.cc/index.php?topic=148850
2. If you haven't already done so, create a new thread in the appropriate section of the forum (http://forum.arduino.cc/), following all the rules. Be sure to post the full error output, using code tags (`</>` button on the toolbar).