arduino/Arduino #9579: [Library Manager] J1850-VPW-Arduino-Transceiver-Library

matafonoff, 2020-01-01:
This is an implementation of J1850VPW protocol for Arduino.

https://github.com/matafonoff/J1850-VPW-Arduino-Transceiver-Library

per1234, 2020-01-01:
That URL gives me a 404 error. Did you maybe forget to make the repository public?

matafonoff, 2020-01-01:
Yep, sorry! Fixed.

matafonoff, 2020-01-05:
@per1234 Any updates here?

per1234, 2020-01-05:
Your library will not be available for installation via Library Manager until it has been added to the index. cmaglie occasionally comes through to clear all the outstanding Library Manager inclusion requests and closes those issues when that is done. You will receive a notification from GitHub of this and your library will be available in Library Manager shortly after that. So we'll get to it soon, please be patient.

matafonoff, 2020-01-06:
@per1234 , thanks for information. Looking forward to it )

cmaglie, 2020-01-07:
Done.

closed 2020-01-07 by cmaglie