arduino/Arduino #9580: OTA port not seen (2.6.3)

dagnall53, 2020-01-01:
Sorry to open a subject that seems to have been solved previously.. But I have read the other issues threads and nothing suggested worked.
I am hoping someone can assist me to track down an issue that started only with an update from 2.5.2 to 2.6.3

In summary: 
Some esp8266 Code I am working with worked fine with 2.5.2, but a recent update to 2.6.3 has stopped the OTA working. I cannot see the esp WiFi port in bonjour browser, or through ping or in a restarted Arduino. Rebooting the router has no effect.

- If I upload the code using 2.5.2, it works fine, the OTA port appears in Arduino etc.. 
- If I update to 2.6.2 the OTA port (in the esp programmed with 2.5.2)  is still visible , (and I can upload using the OTA) but as soon as I do, the port disappears again when Arduino is reloaded. 
- With 2.6.3 I can use the ArduinoOTA example and it works - I have even added things like SPIFFs and ftp to the example and it still works.  
- The code I am working with is very complex and uses AP, Webserver, SPIFFs and OTA.  

From two days worth of experiments I can say definitely that It is just the (very complex) code I am working with that has the incompatibility between 2.5.2 and 2.6.3 and also that I have not managed to make the ArduinoOTA example break by adding SPIFFs or ftp. (but I have not tried adding a webserver or AP. ) .. SO I am asking for help tracking down the incompatibility.. I have tried setting OTA debug via serial, and got these messages: To me it looks like the OTA should be working, but nothing sees the port.. 
Perhaps the debug messages may assist? Obviously there is OTA debug stuff interspersed with my own serial print comments.. 
They look VERY similar to me, but perhaps they may hint at the problem ..
Are there other debug settings that can give me more information? 
Many thanks
Dagnall 

(**with non OTA working 2.6.3** ) 
SDK:2.2.2-dev(38a443e)/Core:2.6.3=20603000/lwIP:STABLE-2_1_2_RELEASE/glue:1.2-16-ge23a07e/BearSSL:89454af

scandone
Connecting to [linksys-25]
WiFi waiting to connect...
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 6
cnt 
WiFi waiting to connect...
WiFi waiting to connect...

connected with linksys-25, channel 11
dhcp client start...
WiFi waiting to connect...
WiFi waiting to connect...
WiFi waiting to connect...
ip:192.168.0.75,mask:255.255.255.0,gw:192.168.0.1
WiFi connected RSSI=-35
-----Starting OTA------
 My ip:192.168.0.75
 My OTA name:WIO33
OTA server at: WIO33.local:8266
-----OTA started------
options 80
I2C device found at address 0x20
I2C device found at address 0x3C
I2C found 2 devices (4,5)
BMP280 status: 0x76=not found 0x77=not found
SSD1306 status: 0x3C=OK 0x3D=not found
bcn 0
del if1
pm open,type:2 0
add if1
pm close 7
dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
bcn 100
AP IP address: 192.168.4.1
Connecting to [192.168.0.25]...
server connected
input status retry :33:0 1

**With the working 2.5.2 version..** 
<SDK:2.2.1(cfd48f3)/Core:2.5.2=20502000/lwIP:STABLE-2_1_2_RELEASE/glue:1.1-7-g82abda3/BearSSL:a143020

scandone
Connecting to [linksys-25]
WiFi waiting to connect...
WiFi waiting to connect...

scandone
state: 0 -> 2 (b0)
WiFi waiting to connect...
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 6
cnt 
WiFi waiting to connect...
WiFi waiting to connect...

connected with linksys-25, channel 11
dhcp client start...
WiFi waiting to connect...
WiFi waiting to connect...
WiFi waiting to connect...
WiFi waiting to connect...
ip:192.168.0.75,mask:255.255.255.0,gw:192.168.0.1
WiFi connected RSSI=-36
-----Starting OTA------
 My ip:192.168.0.75
 My OTA name:WIO33
OTA server at: WIO33.local:8266
-----OTA started------
options 80
I2C device found at address 0x20
I2C device found at address 0x3C
I2C found 2 devices (4,5)
BMP280 status: 0x76=not found 0x77=not found
SSD1306 status: 0x3C=OK 0x3D=not found
bcn 0
del if1
pm open,type:2 0
add if1
pm close 7
dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
bcn 100
AP IP address: 192.168.4.1
Connecting to [192.168.0.25]...
server connected
input status retry :33:0 1
>

facchinm, 2020-01-02:
Hi @dagnall53 ,
you should report this problem at https://github.com/esp8266/Arduino since it looks like it's just dependant on esp8266 core version.

closed 2020-01-02 by facchinm

dagnall53, 2020-01-02:
Yes, I spotted that late last night :-(   
Now started again as https://github.com/esp8266/Arduino/issues/6975