arduino/Arduino #9582: [Library Manager] Add three new libraries

ardnew, 2020-01-02:
Please add the following three libraries to the Library Manager:

---

https://github.com/ardnew/ILI9341-Layout-Manager

**Description**: A modal framework for presenting collections of data using a scaling autolayout engine for ILI9341 chipsets
**Latest release package**: https://github.com/ardnew/ILI9341-Layout-Manager/releases/tag/v1.0.0

---

https://github.com/ardnew/STUSB4500

**Description**: I2C device driver library for USB Power Delivery (PD) sink controller (STUSB4500)
**Latest release package**: https://github.com/ardnew/STUSB4500/releases/tag/v1.0.0

---

https://github.com/ardnew/XPT2046_Calibrated

**Description**: Fork of `XPT2046_Touchscreen` library adding three-point calibration
**Latest release package**: https://github.com/ardnew/XPT2046_Calibrated/releases/tag/v1.0.0

ardnew, 2020-01-05:
As I make updates to these libraries before they get added to the library manager, do I just create a new tag/release and update the reference in this issue?

per1234, 2020-01-05:
We only need to know the repository URLs. The Arduino Library Manager indexer periodically scans all repository URLs in the index and picks up all tags that meet the [requirements for inclusion](https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ).

So just create a new tag/release whenever the code is ready for it. They will all be available via Library Manager within a couple hours after your URLs are added to the list. No need to edit this issue every time or create a new issue.

cmaglie, 2020-01-07:
There is alread a library called `XPT2046_Touchscreen`, to fix this please:
1) change the name in `library.properties`
2) commit & push
3) delete and redo the release tag

The other two libraries have been added.

ardnew, 2020-01-07:
@cmaglie 
The name of the library in `library.properties` for the latest tagged release is `XPT2046_Calibrated` not `XPT2046_Touchscreen`.

It is a fork of the project `XPT2046_Touchscreen`, but `library.properties` was already updated. Something else I need to change?

cmaglie, 2020-01-08:
ah, ok now I see it, I'll add as `XPT2046_Calibrated` then, the other tags will be ignored.

cmaglie, 2020-01-08:
Done.

closed 2020-01-08 by cmaglie