arduino/Arduino #9584: Please add this library

navintea, 2020-01-03:
[https://github.com/Protocentral/ADS1292rShield_Breakout/tree/master/software/arduino_library/Protocentral_ADS1292R]

per1234, 2020-01-03:
Hi @navintea There are a couple of problems that will prevent this library from being added to the Library Manager index as is:
- The library must be in the root of the repository, not in the `software/arduino_library/Protocentral_ADS1292R/` subfolder, as it is now.
- The library must have a library.properties file.

If you still want the library added, please make the necessary changes for compliance with the Arduino Library Manager requirements and then create a new release/tag. After that, comment here so we'll know we can proceed.

More information:
- https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ
- https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5:-Library-specification#libraryproperties-file-format

navintea, 2020-01-03:
Hello,
          Surely will look into the changes to be made, where can i
lookup for more detailed info of what you've mentioned anywhere else?.
Thanks for the guidance.

Thanks & Regards,

Navin

per1234, 2020-01-03:
All the information you need is at the two links I posted in my previous reply. If you follow those faithfully, your library will be compliant.

If you want to go farther than that, I have a script you can use to check compliance with the requirements for inclusion in the Library Manager index:
https://github.com/per1234/ino-library-manager-validator
Probably more effort than it's worth to set up for a single library, but there it is.

Possibly more useful to you is the `check_library_manager_compliance` function of this script:
https://github.com/per1234/arduino-ci-script#check_library_manager_compliance-librarypath
That can be used in your continuous integration tests to make sure the library's compliance isn't later inadvertently broken during development.

cmaglie, 2020-02-17:
I'm closing this for now. Please, open a new request when the library is ready.

closed 2020-02-17 by cmaglie