arduino/Arduino #9585: [Library Manager] add SunEzy / Orion monitoring library to library manger

guglicap, 2020-01-03:
Hi, 
I wrote this library to communicate with SunEzy / Orion inverters, I was wondering if you could add it. 
Please let me know if a name change is needed because although it doesn't start with "Arduino" the name is "arduino-pv". Thanks.

github.com/guglicap/arduino-pv

per1234, 2020-01-03:
Hi @guglicap The restriction on library names starting with "Arduino" applies to any capitalization of the word. If you still want to add you library to the Arduino Library Manager index, you'll need to change the `name` value in library.properties, then either create a new release/tag (after updating the `version` value in library.properties), or redo the existing release/tag.

After doing so, please comment here so we'll know we can proceed.

guglicap, 2020-01-03:
done, thanks for the quick reply.

cmaglie, 2020-01-07:
Done.

closed 2020-01-07 by cmaglie