arduino/Arduino #9590: Win32 Arduino IDE 1.8.10: library not found after installing LLVM for Windows

neographikal, 2020-01-05:
I had a working .ino solution, but since I wanted code completion in Visual Studio code I had to install LLVM for Windows for an extension in VSCode. After installing LLVM (and uninstalling it after the builds failed) I can't build the .ino anymore. I commented all code and started adding code one by one.

```
#include "PubSubClient.h" // works
#include "WiFi.h" // works
#include <CAN_config.h> // does not work
```
Error:

```
panic: runtime error: invalid memory address or nil pointer dereference
[signal 0xc0000005 code=0x0 addr=0x0 pc=0x7855fb]

goroutine 1 [running]:
github.com/arduino/arduino-cli/legacy/builder.ResolveLibrary(0x11ce85a0, 0x11c92876, 0xc, 0x11c92876)
	/home/jenkins/workspace/arduino-builder-all-cross-cli-inception/src/github.com/arduino/arduino-cli/legacy/builder/resolve_library.go:64 +0x14b
github.com/arduino/arduino-cli/legacy/builder.findIncludesUntilDone(0x11ce85a0, 0x11d3cf20, 0x88ed00, 0x11c98e40, 0x11f6a210, 0x11c53a00, 0x0)
	/home/jenkins/workspace/arduino-builder-all-cross-cli-inception/src/github.com/arduino/arduino-cli/legacy/builder/container_find_includes.go:358 +0x3b6
github.com/arduino/arduino-cli/legacy/builder.(*ContainerFindIncludes).Run(0xdebc68, 0x11ce85a0, 0xdebc68, 0x0)
	/home/jenkins/workspace/arduino-builder-all-cross-cli-inception/src/github.com/arduino/arduino-cli/legacy/builder/container_find_includes.go:152 +0x3e5
github.com/arduino/arduino-cli/legacy/builder.runCommands(0x11ce85a0, 0x11c3be1c, 0x20, 0x20, 0x409a01, 0x3f, 0x0)
	/home/jenkins/workspace/arduino-builder-all-cross-cli-inception/src/github.com/arduino/arduino-cli/legacy/builder/builder.go:210 +0xbd
github.com/arduino/arduino-cli/legacy/builder.(*Builder).Run(0x11c79f40, 0x11ce85a0, 0xdebbb4, 0x11c4ac90)
	/home/jenkins/workspace/arduino-builder-all-cross-cli-inception/src/github.com/arduino/arduino-cli/legacy/builder/builder.go:117 +0xb9c
github.com/arduino/arduino-cli/legacy/builder.RunBuilder(...)
	/home/jenkins/workspace/arduino-builder-all-cross-cli-inception/src/github.com/arduino/arduino-cli/legacy/builder/builder.go:226
main.main()
	/home/jenkins/workspace/arduino-builder-all-cross-cli-inception/src/github.com/arduino/arduino-builder/main.go:398 +0x6d2
C:\Users\neo\Desktop\arduino-1.8.10\arduino-builder returned 2
Error compiling for board ESP32 Dev Module.
```

[image]

File is and was present, did build without problems. I'm confused :)

per1234, 2021-03-28:
Hi @ict-one-nl. Thanks so much for reporting this.

I wanted to check in with you to find out whether you are still able to reproduce this with the latest hourly build of the Arduino IDE:
https://www.arduino.cc/en/software#hourly-builds

There have been quite a few advancements in Arduino CLI (which is now the base of arduino-builder) since the time of your report so its possible one of those things has resolved this issue.

ict-one-nl, 2021-03-31:
Sorry switched to a mac now, can't reproduce anymore :(

closed 2021-03-31 by per1234