arduino/Arduino #9595: Please add the ACS-M1128 Library

danielcy77, 2020-01-06:
https://github.com/samelement/ACS-M1128

per1234, 2020-01-06:
Hi @danielcy77 Although the library currently does meet the requirements for inclusion in the Arduino Library Manager index, it did not at the time of the [last tag/release](https://github.com/samelement/ACS-M1128/tree/v2.2.1-b1). At that time, it was missing the required library.properties file. We won't be able to add your library to the list until the repository contains a tag/release that meets the Library Manager requirements.

So please create a new Git tag or GitHub release. Once that's done, comment here so we'll know we can proceed.

danielcy77, 2020-01-07:
@per1234 
We have released a new version of this library. And library.properties file has been added in it.Can this library be processed? 

link tree: https://github.com/samelement/ACS-M1128/tree/2.2.1-b2

per1234, 2020-01-07:
@danielcy77 looks good. Thanks!

cmaglie, 2020-01-07:
Done.

closed 2020-01-07 by cmaglie